gqrx-sdr/gqrx #730: can not run gqrx. gnuradio3.8 run on my computer, but gqrx said can not find so.3.7.11

jin-github1002, 2020-01-31:
hello, i installed gqrx from source. when i make it. There are errors about gnuradio3.7. but gnuradio3.8 runs on my computer. why is it? how to deal with it?

dell@dell-XPS-15-9550:~/gqrx.git/build$ cmake ..
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gnuradio-osmosdr'
--   Found gnuradio-osmosdr, version 0.1.4
-- Found GNURADIO_OSMOSDR: /usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so  
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.7
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Checking for module 'alsa'
--   Found alsa, version 1.1.3
-- Found ALSA 1.1.3
-- Checking for module 'jack'
--   Found jack, version 0.125.0rc1
-- Found jack: /usr/lib/x86_64-linux-gnu/libjack.so
-- Checking for module 'portaudio-2.0'
--   Found portaudio-2.0, version 19
-- Performing Test PORTAUDIO2_FOUND
-- Performing Test PORTAUDIO2_FOUND - Success
-- Found PORTAUDIO: /usr/include  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dell/gqrx.git/build
[... 89 lines cut ...]
/usr/bin/ld: warning: libgnuradio-iqbalance.so.3.7.11, needed by /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libgnuradio-fcd.so.3.7.11, needed by /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libgnuradio-fcdproplus.so.3.7.11, needed by /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libgnuradio-uhd.so.3.7.11, needed by /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so, not found (try using -rpath or -rpath-link)
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `pmt::dict_has_key(boost::intrusive_ptr<pmt::pmt_base> const&, boost::intrusive_ptr<pmt::pmt_base> const&)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `gr::msg_accepter::post(boost::intrusive_ptr<pmt::pmt_base>, boost::intrusive_ptr<pmt::pmt_base>)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `pmt::intrusive_ptr_add_ref(pmt::pmt_base*)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `gr::iqbalance::fix_cc::make(float, float)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `gr::blocks::file_source::make(unsigned long, char const*, bool)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `pmt::eqv(boost::intrusive_ptr<pmt::pmt_base> const&, boost::intrusive_ptr<pmt::pmt_base> const&)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `gr::fcd::source_c::make(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `pmt::symbol_to_string[abi:cxx11](boost::intrusive_ptr<pmt::pmt_base> const&)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `pmt::intrusive_ptr_release(pmt::pmt_base*)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `gr::fcdproplus::fcdproplus::make(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `pmt::list_has(boost::intrusive_ptr<pmt::pmt_base>, boost::intrusive_ptr<pmt::pmt_base> const&)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `gr::iqbalance::optimize_c::make(int)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `gr::uhd::usrp_sink::make(uhd::device_addr_t const&, uhd::stream_args_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libgnuradio-osmosdr.so: undefined reference to `gr::uhd::usrp_source::make(uhd::device_addr_t const&, uhd::stream_args_t const&, bool)'
collect2: error: ld returned 1 exit status
src/CMakeFiles/gqrx.dir/build.make:1659: recipe for target 'src/gqrx' failed
make[2]: *** [src/gqrx] Error 1
CMakeFiles/Makefile2:135: recipe for target 'src/CMakeFiles/gqrx.dir/all' failed
make[1]: *** [src/CMakeFiles/gqrx.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

alexf91, 2020-01-31:
Your version of gnuradio-osmosdr is not built for GNU Radio 3.8

jin-github1002, 2020-01-31:
now i run gqrx.  there is no soapy in the device colomn. what to do?

csete, 2020-02-02:
I suggest you try a support forum, see for instance https://groups.google.com/forum/#!forum/gqrx
We use github issues for development only and not for user support.

closed 2020-02-02 by csete