gqrx-sdr/gqrx #755: Error compiling dsp/lpf.cpp

gvanem, 2020-02-19:
Compiling `dsp/lpf.cpp` using clang-cl (x86), I'm getting this error I do not understand:
```c
src/dsp/lpf.cpp(61,51): error: variable type 'gr::filter::fir_filter_blk<float, float, float>' is an abstract class
    gr::filter::fir_filter_blk<float,float,float> f_blk;
                                                  ^
../gnuradio/MSVC-build/include\gnuradio/sync_block.h(40,17): note: unimplemented pure virtual method 'work' in 'fir_filter_blk'
    virtual int work(int noutput_items,
                ^
../gnuradio/MSVC-build/include\gnuradio/filter/fir_filter_blk.h(59,18): note: unimplemented pure virtual method 'set_taps' in
      'fir_filter_blk'
    virtual void set_taps(const std::vector<TAP_T>& taps) = 0;
                 ^
../gnuradio/MSVC-build/include\gnuradio/filter/fir_filter_blk.h(60,32): note: unimplemented pure virtual method 'taps' in 'fir_filter_blk'
    virtual std::vector<TAP_T> taps() const = 0;
                               ^
1 error generated.
```
I do have the latest GnuRadio from Github and compile with `-DGNURADIO_VERSION=0x030800`.

With my lack of C++ knowledge, I fail to understand this error.

gvanem, 2020-02-19:
*Edit:* I was able to compile this file with this patch:
```diff
--- a/src/dsp/lpf.cpp 2017-12-14 13:53:07
+++ b/src/dsp/lpf.cpp 2020-02-19 12:17:43
@@ -58,7 +58,12 @@
                                  d_cutoff_freq, d_trans_width);

     /* create low-pass filter (decimation=1) */
-    lpf = gr::filter::fir_filter_fff::make(1, d_taps);
+#if 0
+    gr::filter::fir_filter_blk<float,float,float> f_blk;
+    lpf = f_blk.make(1, d_taps);
+#else
+    lpf = gr::filter::fir_filter_blk<float,float,float>::make(1, d_taps);
+#endif

     /* connect filter */
     connect(self(), 0, lpf, 0);
```

(courtesy of how [GNSS](https://github.com/gnss-sdr/gnss-sdr) implemented a low-pass filter). I've no idea if it works.

kolbma, 2020-03-12:
@gvanem 

Are you sure you are compiling against this gnuradio version...

The problem is related to this condition
https://github.com/csete/gqrx/blob/7f0c1d92f5de3a52d778004001be6f0d5d5d96f7/src/dsp/lpf.h#L29-L33

In `gnuradio/filter/fir_filter_blk.h` 
there is a 
```
namespace gr {
namespace filter {
typedef fir_filter_blk<float, float, float> fir_filter_fff;
```

So

https://github.com/csete/gqrx/blob/7f0c1d92f5de3a52d778004001be6f0d5d5d96f7/src/dsp/lpf.cpp#L61

**should become exactly the same** you are using in your fix... this line is enough... don't know why you put in complete preprocessor 0 (false) condition.

`lpf = gr::filter::fir_filter_blk<float,float,float>::make(1, d_taps);`

Maybe you can have a look at what your preprocessor outputs...
https://clang.llvm.org/docs/ClangCommandLineReference.html#dumping-preprocessor-state

kolbma, 2020-03-12:
I've just built with clang (but Linux) and the release version 0x030800.
No problem.

argilo, 2020-11-28:
I'll close this off, since CI builds show that Gqrx is compiling fine with GNU Radio 3.7, 3.8 and 3.9.

closed 2020-11-28 by argilo