gqrx-sdr/gqrx #757: usb_claim_interface error because osmosdr devices are not closed

kolbma, 2020-02-20:
The osmosdr devices are closed in the destructor.
Because in gqrx there is a `receiver` singleton with device-`src` instance object (shared_ptr), this destructor is only called on program exit.

So it is not possible to reconfigure the osmosdr input device without restarting gqrx.

A pull request with fix is on the way...

kolbma, 2020-02-21:
The commit https://github.com/kolbma/gqrx/commit/66b123fe6ffe20c667d0b56d2167b76702d383a6

makes https://github.com/csete/gqrx/pull/751 obsolete.

I handle this with an existing dialog.

kolbma, 2020-02-21:
If the usb_claim_interface error still occurs and switching to a different device configuration doesn't work correctly, you can start the SDR in this state (doesn't receive anything) and stop it. Afterwards try the device configuration again. It should work. 
I think this is a bug in gnuradio, but maybe someone can have a look at, why the osmosdr `src` device will be only closed in this case when `tb->start()` is called?!

kolbma, 2020-02-21:
I've patched my gnuradio with https://github.com/gnuradio/gnuradio/pull/3184
Then it always works.

argilo, 2020-02-21:
> Because in gqrx there is a `receiver` singleton with device-`src` instance object (shared_ptr), this destructor is only called on program exit.

The shared pointer is reset here:

https://github.com/csete/gqrx/blob/490ee3be4f1c5e0319b7610912104e2d0a47fd4e/src/applications/gqrx/receiver.cpp#L229

This should cause the device's destructor to be called. The only reason it isn't called is that GNU Radio holds onto a reference to it unnecessarily. I corrected that in https://github.com/gnuradio/gnuradio/pull/3184, and the fix should be backported to GNU Radio 3.8 in https://github.com/gnuradio/gnuradio/pull/3214.

csete, 2020-02-22:
How can I reproduce this error?

argilo, 2020-02-22:
To reproduce:

1. Make sure your GNU Radio is not yet patched with https://github.com/gnuradio/gnuradio/pull/3184 or https://github.com/gnuradio/gnuradio/pull/3214.
2. Open Gqrx with an RTL-SDR and click the "Start DSP" button.
3. Stop DSP processing.
4. Open the devices dialog and change the RTL-SDR's device string (for example, adding `buflen=32768`.
5. Click OK.

At this point, Gqrx will be unable to re-open the RTL-SDR because GNU Radio has held onto a reference to the osmocom source block, keeping the USB interface open.

csete, 2020-02-22:
I see, thanks.

argilo, 2020-02-22:
By the way, the GNU Radio folks agree that this is a bug in GNU Radio, but they also pointed out that gr-osmosdr should release connections to hardware in `stop()` rather than the block destructor. That's a bigger change, so I haven't undertaken it yet.

kolbma, 2020-02-23:
Ok. Then should I do 2 pull requests...?
1. With the mis-config and no-device handling (without vector)
2. The direct sampling config option for rtl devices

argilo, 2020-02-23:
I don't believe 1 is necessary, since the issues were already fixed in https://github.com/gnuradio/gnuradio/pull/3184 and #751.

It would make sense to put 2 into its own pull request. I'm not sure adding a second copy of the device to the device list for that is the right approach though -- maybe there should just be a checkbox for that option somewhere in the device dialog?

kolbma, 2020-02-23:
But you handle it in #751 with a required app restart.
This is now without vector...
[gqrx-device-set-crash.txt](https://github.com/csete/gqrx/files/4241751/gqrx-device-set-crash.txt)

argilo, 2020-02-23:
> But you handle it in #751 with a required app restart.

That's not correct; there's no need to restart Gqrx.

kolbma, 2020-02-24:
The device list is only updated once during start here https://github.com/csete/gqrx/blob/7f0c1d92f5de3a52d778004001be6f0d5d5d96f7/src/applications/gqrx/mainwindow.cpp#L298

So you have to restart to get the correct device list.

csete, 2020-02-24:
> I'm not sure adding a second copy of the device to the device list for that is the right approach though -- maybe there should just be a checkbox for that option somewhere in the device dialog?

I think it's a bad idea to have device-specific code in gqrx when we are using a device abstraction library. Different devices have many useful settings that can only be set through device string arguments. Therefore, it would be much better if a list of supported device string arguments could be retrieved from gr-osmosdr and supported through the device control panel.

kolbma, 2020-02-24:
@csete 
I see it the same way, but this is some work to refactor the whole configuration and device handling in an OO fashion. I just want to get it to work without SEGVs or restarting the app.
And typing in my direct_samp=2 over and over again when changing frequency areas.

argilo, 2020-02-24:
> The device list is only updated once during start here

OK, now I understand what you mean. I also find it a bit annoying that you have to restart Gqrx if you plug in a new device. I considered changing that, but this comment about device probing potentially changing device configuration gave me pause:

https://github.com/csete/gqrx/blob/7f0c1d92f5de3a52d778004001be6f0d5d5d96f7/src/applications/gqrx/mainwindow.cpp#L296-L298

I didn't yet check whether that's still an issue.

kolbma, 2020-02-26:
Well, if you have a running gqrx and unplug the usb device. Then later you plugin the usb device again. You don't get the device to work in gqrx. 
DSP processing starts, but there is no radio action.

It needs a restart of gqrx. Annoying.

kolbma, 2020-03-05:
I close this.
There is a pull https://github.com/csete/gqrx/pull/774

closed 2020-03-05 by kolbma