gqrx-sdr/gqrx #766: Preset needs to be filter and not filtershape

kolbma, 2020-03-02:
https://github.com/csete/gqrx/blob/7f0c1d92f5de3a52d778004001be6f0d5d5d96f7/src/applications/gqrx/mainwindow.cpp#L2091

https://github.com/csete/gqrx/blob/7f0c1d92f5de3a52d778004001be6f0d5d5d96f7/src/applications/gqrx/mainwindow.cpp#L2117

should be

```int preset = uiDockRxOpt->currentFilter();```

alexf91, 2020-03-03:
Nice catch, feel free to create a pull request.
The logic around filter settings is very buggy. That's certainly something that deserves improvements.

Just out of interest: How did you find this? It took me almost half an hour to figure out if this report was correct.

kolbma, 2020-03-04:
Hi Alex,
because I got strange behavior during testing for with https://github.com/csete/gqrx/issues/764 and couldn't find any reason in my code.

It would be fixed in the corresponding pull. But travis fails at the moment. I'm not sure if the old Qt version or the old GCC version is responsible for the fail.

Would be nice to know which API version we should develop for...

csete, 2020-04-16:
Fixed, thanks @kolbma

closed 2020-04-16 by csete