gqrx-sdr/gqrx #777: gqrx 2.11.5 cannot build with gnuradio 3.8.0

hobbes1069, 2020-03-20:
It looks like gnuradio removed/changed some headers with the 3.8.0 release and is now uninstallable on Fedora 32 (soon to be released).

```
g++ -c -pipe -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -DVERSION="\"2.11.5\"" -DWITH_PULSEAUDIO -D_REENTRANT -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -Isrc -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtSvg -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtCore -I. -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o remote_control_settings.o src/applications/gqrx/remote_control_settings.cpp
In file included from src/applications/gqrx/file_resources.cpp:24:
src/applications/gqrx/receiver.h:26:10: fatal error: gnuradio/analog/sig_source_c.h: No such file or directory
   26 | #include <gnuradio/analog/sig_source_c.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:1052: file_resources.o] Error 1
make: *** Waiting for unfinished jobs....
src/applications/gqrx/receiver.cpp:31:10: fatal error: gnuradio/blocks/multiply_const_ff.h: No such file or directory
   31 | #include <gnuradio/blocks/multiply_const_ff.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:1033: receiver.o] Error 1
In file included from src/applications/gqrx/mainwindow.h:49,
                 from src/applications/gqrx/main.cpp:42:
src/applications/gqrx/receiver.h:26:10: fatal error: gnuradio/analog/sig_source_c.h: No such file or directory
   26 | #include <gnuradio/analog/sig_source_c.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:963: main.o] Error 1
In file included from src/applications/gqrx/mainwindow.h:49,
                 from src/applications/gqrx/mainwindow.cpp:49:
src/applications/gqrx/receiver.h:26:10: fatal error: gnuradio/analog/sig_source_c.h: No such file or directory
   26 | #include <gnuradio/analog/sig_source_c.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:1006: mainwindow.o] Error 1
src/applications/gqrx/remote_control.cpp: In member function 'void RemoteControl::startAudioRecorder(QString)':
src/applications/gqrx/remote_control.cpp:352:48: warning: unused parameter 'unused' [-Wunused-parameter]
  352 | void RemoteControl::startAudioRecorder(QString unused)
      |                                        ~~~~~~~~^~~~~~
```

https://bugzilla.redhat.com/show_bug.cgi?id=1815481

csete, 2020-03-20:
This has been fixed and there is a release tag for 2.12.1, please use that.

closed 2020-03-20 by csete

hobbes1069, 2020-03-20:
So far so good, thanks!