gqrx-sdr/gqrx #779: NFM demodulation issue

antonblanchard, 2020-04-09:
I'm having issues with NFM demodulation since a1fc84feef9f ('app/gqrx: Replace signal_source + multiply with rotator').

I can still hear the audio, but it sounds like it's being heavily clipped and there is a loud buzz.

argilo, 2020-04-09:
@antonblanchard Which version/commit of GNU Radio are you using? I suspect you might be hitting this bug: https://github.com/gnuradio/gnuradio/issues/2748

antonblanchard, 2020-04-10:
Thanks @argilo, that is indeed the issue. I was building inside a Fedora 32 container, which has gnuradio 3.8.0.0. I've submitted a Fedora bug at:

https://bugzilla.redhat.com/show_bug.cgi?id=1822842

argilo, 2020-10-18:
I'll close this off, since the bug was not in Gqrx.

closed 2020-10-18 by argilo