gqrx-sdr/gqrx #781: gqrx crashes when initializing dsp (exotic architecture)

CodeIsAnEnigma, 2020-04-13:
Hello,
I'm having an issue involving gqrx, where I get a segmentation fault (core dumped) error any time I try to initialize my SDR. Here is the background on my current setup...

1. I'm currently running on Fedora 31 on ppc64le architecture
2. I have a nuand bladeRF 2.0 connected via a SuperSpeed xHCI USB port
3. The bladeRF utilities detect the SDR with no issues
4. I compiled GNU Radio 3.8, gr-osmosdr, and the latest gqrx from source

I have been able to start up gqrx successfully and select the bladeRF device from the drop down list in the configure I/O devices dialog box. However, upon starting the DSP, the entire program crashes and I get the following text in terminal...

`[jsn@hamshack-ppc64le ~]$ gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl_tcp bladerf rfspace redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
BookmarksFile is /home/jsn/.config/gqrx/bookmarks.csv
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl_tcp bladerf rfspace redpitaya 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # 42b3...5483 FW v2.3.2 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
FOUND full
[bladeRF source] start: DEBUG: starting source
Segmentation fault (core dumped)`

Could there be an additional library I need to compile for my specific architecture that is normally "assumed" for x86_64 style set ups? Is there a way I can present more detailed and meaningful detail for research purposes? The error is a bit vague, and other solutions to this kind of error don't seem to be applicable to my situation. Any advice is welcomed!

csete, 2020-04-14:
Hi,
I don't think you need any additional libraries as such things would prevent gqrx from starting at all. Could you try to run gqrx through gdb and type `bt` in the gdb shell after gqrx crashes? THat might give a hint where it crashes, although it might be necessary to rebuild gqrx and the libraries with debug symbols. But just try it first.

CodeIsAnEnigma, 2020-04-14:
Hope this helps. I recompiled with debug symbols enabled.

> (gdb) run
Starting program: /usr/local/bin/gqrx 
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.30-11.fc31.ppc64le
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffe38ef050 (LWP 7231)]
[New Thread 0x7fffdbfff050 (LWP 7232)]
[New Thread 0x7fffdb7ef050 (LWP 7233)]
[Thread 0x7fffdb7ef050 (LWP 7233) exited]
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl_tcp bladerf rfspace redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
[New Thread 0x7fffdb7ef050 (LWP 7234)]
Using audio backend: auto
BookmarksFile is /home/jsn/.config/gqrx/bookmarks.csv
[New Thread 0x7fffd926f050 (LWP 7235)]
[Thread 0x7fffd926f050 (LWP 7235) exited]
[New Thread 0x7fffd926f050 (LWP 7236)]
[New Thread 0x7fffb3fff050 (LWP 7237)]
[New Thread 0x7fffb37ef050 (LWP 7238)]
[New Thread 0x7fffb2fdf050 (LWP 7239)]
[New Thread 0x7fffb27cf050 (LWP 7240)]
[New Thread 0x7fffb1fbf050 (LWP 7241)]
[New Thread 0x7fffb17af050 (LWP 7242)]
[New Thread 0x7fffb0f9f050 (LWP 7243)]
[New Thread 0x7fff93fff050 (LWP 7244)]
[New Thread 0x7fff8bfff050 (LWP 7245)]
[New Thread 0x7fff937ef050 (LWP 7246)]
[New Thread 0x7fff92fdf050 (LWP 7247)]
[New Thread 0x7fff927cf050 (LWP 7248)]
[New Thread 0x7fff91fbf050 (LWP 7249)]
[New Thread 0x7fff917af050 (LWP 7250)]
[New Thread 0x7fff90f9f050 (LWP 7251)]
[New Thread 0x7fff8b7ef050 (LWP 7252)]
[New Thread 0x7fff8afdf050 (LWP 7253)]
[New Thread 0x7fff8a7cf050 (LWP 7254)]
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl_tcp bladerf rfspace redpitaya 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[New Thread 0x7fff89fbf050 (LWP 7255)]
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # 42b3...5483 FW v2.3.2 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
IQ DCR samp_rate: 3.072e+07
IQ DCR alpha: 3.25521e-08
Changing NB_RX quad rate: 96000 -> 3.072e+07
FOUND full
No change in output device:
  old: 
  new: 
IQ DCR samp_rate: 5e+06
IQ DCR alpha: 2e-07
Changing NB_RX quad rate: 3.072e+07 -> 5e+06
Generating taps for new filter   LO:-5000   HI:5000 TW:2000   Taps: 115
Generating taps for new filter   LO:-5000   HI:5000 TW:2000   Taps: 115
[New Thread 0x7fff8900f050 (LWP 7256)]
[bladeRF source] start: DEBUG: starting source
[New Thread 0x7fff619bf050 (LWP 7257)]
[New Thread 0x7fff611af050 (LWP 7258)]
[New Thread 0x7fff6099f050 (LWP 7259)]
[New Thread 0x7fff4ffff050 (LWP 7260)]
[New Thread 0x7fff4efdf050 (LWP 7262)]
[New Thread 0x7fff4f7ef050 (LWP 7261)]
[New Thread 0x7fff4e7cf050 (LWP 7263)]
[... 15 lines cut ...]
0x00007ffff5545a50 in volk_32fc_s32fc_x2_rotator_32fc_generic ()
   from /usr/local/lib/libvolk.so.2.2.1
Missing separate debuginfos, use: dnf debuginfo-install OpenEXR-libs-2.3.0-4.fc31.ppc64le alsa-lib-1.2.2-1.fc31.ppc64le boost-atomic-1.69.0-9.fc31.ppc64le boost-chrono-1.69.0-9.fc31.ppc64le boost-date-time-1.69.0-9.fc31.ppc64le boost-filesystem-1.69.0-9.fc31.ppc64le boost-program-options-1.69.0-9.fc31.ppc64le boost-regex-1.69.0-9.fc31.ppc64le boost-system-1.69.0-9.fc31.ppc64le boost-test-1.69.0-9.fc31.ppc64le boost-thread-1.69.0-9.fc31.ppc64le boost-timer-1.69.0-9.fc31.ppc64le bzip2-libs-1.0.8-1.fc31.ppc64le dbus-libs-1.12.16-3.fc31.ppc64le dbusmenu-qt5-0.9.3-0.21.20160218.fc31.ppc64le elfutils-libelf-0.179-1.fc31.ppc64le expat-2.2.8-1.fc31.ppc64le fftw-libs-single-3.3.8-6.fc31.ppc64le flac-libs-1.3.3-1.fc31.ppc64le fontconfig-2.13.92-3.fc31.ppc64le freetype-2.10.0-3.fc31.ppc64le glib2-2.62.6-1.fc31.ppc64le gmp-6.1.2-10.fc31.ppc64le gmp-c++-6.1.2-10.fc31.ppc64le graphite2-1.3.13-1.fc31.ppc64le gsm-1.0.18-5.fc31.ppc64le harfbuzz-2.6.1-2.fc31.ppc64le ilmbase-2.3.0-3.fc31.ppc64le jasper-libs-2.0.14-9.fc31.ppc64le jbigkit-libs-2.1-17.fc31.ppc64le kf5-attica-5.68.0-1.fc31.ppc64le kf5-frameworkintegration-libs-5.68.0-1.fc31.ppc64le kf5-karchive-5.68.0-1.fc31.ppc64le kf5-kauth-5.68.0-1.fc31.ppc64le kf5-kbookmarks-5.68.0-1.fc31.ppc64le kf5-kcodecs-5.68.0-1.fc31.ppc64le kf5-kcompletion-5.68.0-1.fc31.ppc64le kf5-kconfig-core-5.68.0-1.fc31.ppc64le kf5-kconfig-gui-5.68.0-1.fc31.ppc64le kf5-kconfigwidgets-5.68.0-1.fc31.ppc64le kf5-kcoreaddons-5.68.0-1.fc31.ppc64le kf5-kcrash-5.68.0-1.fc31.ppc64le kf5-kdbusaddons-5.68.0-1.fc31.ppc64le kf5-kglobalaccel-libs-5.68.0-1.fc31.ppc64le kf5-kguiaddons-5.68.0-1.fc31.ppc64le kf5-ki18n-5.68.0-1.fc31.ppc64le kf5-kiconthemes-5.68.0-1.fc31.ppc64le kf5-kimageformats-5.68.0-1.fc31.ppc64le kf5-kio-core-libs-5.68.0-1.fc31.ppc64le kf5-kio-file-widgets-5.68.0-1.fc31.ppc64le kf5-kio-widgets-libs-5.68.0-1.fc31.ppc64le kf5-kitemviews-5.68.0-1.fc31.ppc64le kf5-kjobwidgets-5.68.0-1.fc31.ppc64le kf5-knotifications-5.68.0-1.fc31.ppc64le kf5-kservice-5.68.0-1.fc31.ppc64le kf5-kwayland-5.68.0-1.fc31.ppc64le kf5-kwidgetsaddons-5.68.0-1.fc31.ppc64le kf5-kwindowsystem-5.68.0-1.fc31.ppc64le kf5-kxmlgui-5.68.0-1.fc31.ppc64le kf5-solid-5.68.0-1.fc31.ppc64le lcms2-2.9-6.fc31.ppc64le libICE-1.0.10-2.fc31.ppc64le libSM-1.2.3-4.fc31.ppc64le libX11-1.6.9-2.fc31.ppc64le libX11-xcb-1.6.9-2.fc31.ppc64le libXau-1.0.9-2.fc31.ppc64le libXcursor-1.1.15-6.fc31.ppc64le libXdamage-1.1.4-17.fc31.ppc64le libXext-1 [... cut ...]
#7  0x00007ffff5661b58 in boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke(boost::detail::function::function_buffer&) ()
   from /usr/local/lib/libgnuradio-runtime.so.3.8.1
#8  0x00007ffff568b288 in boost::detail::thread_data<boost::function0<void> >::run() () from /usr/local/lib/libgnuradio-runtime.so.3.8.1
#9  0x00007ffff7921d38 in thread_proxy ()
--Type <RET> for more, q to quit, c to continue without paging--
   from /lib64/libboost_thread.so.1.69.0
#10 0x00007ffff78c9a88 in start_thread () from /lib64/libpthread.so.0
#11 0x00007ffff4c139b8 in clone () from /lib64/libc.so.6

csete, 2020-04-16:
Indeed, that shows clearly where the error occurs.
I wonder whether it is still the same problem as encountered in issue #779 or something new. I was under the impression that the issue was fixed with the latest GNU Radio and Volk. Could you try adding `export VOLK_GENERIC=1` (assuming you are using bash) before running gqrx? Thanks.

argilo, 2020-04-16:
From the stack trace it looks like the generic rotator kernel that's getting used. (I don't expect any others are available on this architecture.) I'm not aware of any bugs in [that kernel](https://github.com/gnuradio/volk/blob/dca9988935a95aa7dda38a5c6407e15af520c5f0/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h#L96-L120), so I'm not sure what's going on here.

If it's possible, it might be helpful to try this in valgrind. Or try building VOLK from source so that line number information becomes available.

CodeIsAnEnigma, 2020-04-16:
Ok so I have good news. When I pass the export variable, my waterfall now shows up in gqrx!

However, I have one other issue. I don't hear anything from my speakers, even though the window suggests audio is porting to them. What common settings should I be cognizant of in order to hear what is being received over the air?

Thank you again for all of your help!

CodeIsAnEnigma, 2020-04-16:
Just as a sanity check, I looked at my audio settings in KDE and gqrx has the volume cranked up to 100%, and audio is set to default to my monitor speakers across the board. I also went into the I/O devices configuration and manually selected my desired audio device instead of leaving it at "Default".

csete, 2020-04-18:
If you set the mode to anything except "Demod Off" and squelch at default -150 dB you should at least hear some noise and see it on the audio spectrum.

CodeIsAnEnigma, 2020-04-20:
Hi csete!

So I tried changing the demod setting, and unfortunately that caused a crash. See below for gdb and bt output...

> Starting program: /usr/local/bin/gqrx 
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.30-11.fc31.ppc64le
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffe38ef050 (LWP 3693)]
[New Thread 0x7fffdbfff050 (LWP 3694)]
[New Thread 0x7fffdb7ef050 (LWP 3695)]
[Thread 0x7fffdb7ef050 (LWP 3695) exited]
[New Thread 0x7fffdb7ef050 (LWP 3696)]
[New Thread 0x7fffce92f050 (LWP 3697)]
[New Thread 0x7fffce11f050 (LWP 3698)]
[New Thread 0x7fffcd90f050 (LWP 3699)]
[New Thread 0x7fffcd0ff050 (LWP 3700)]
[New Thread 0x7fffcc8ef050 (LWP 3701)]
[New Thread 0x7fffb3fff050 (LWP 3702)]
[New Thread 0x7fffab7ef050 (LWP 3703)]
[New Thread 0x7fffb37ef050 (LWP 3704)]
[New Thread 0x7fffb2fdf050 (LWP 3705)]
[New Thread 0x7fffb27cf050 (LWP 3706)]
[New Thread 0x7fffb1fbf050 (LWP 3707)]
[New Thread 0x7fffb17af050 (LWP 3708)]
[New Thread 0x7fffb0f9f050 (LWP 3709)]
[New Thread 0x7fffabfff050 (LWP 3710)]
[New Thread 0x7fffaafdf050 (LWP 3711)]
[New Thread 0x7fffaa7cf050 (LWP 3712)]
[New Thread 0x7fffa9fbf050 (LWP 3713)]
[New Thread 0x7fffa97af050 (LWP 3714)]
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl_tcp bladerf rfspace redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
[New Thread 0x7fffa88af050 (LWP 3715)]
Using audio backend: auto
BookmarksFile is /home/jsn/.config/gqrx/bookmarks.csv
[New Thread 0x7fff62e2f050 (LWP 3716)]
[Thread 0x7fff62e2f050 (LWP 3716) exited]
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl_tcp bladerf rfspace redpitaya 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[New Thread 0x7fff62e2f050 (LWP 3717)]
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # 42b3...5483 FW v2.3.2 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
IQ DCR samp_rate: 3.072e+07
IQ DCR alpha: 3.25521e-08
Changing NB_RX quad rate: 96000 -> 3.072e+07
FOUND full
New audio output device:
   old: 
   new: alsa_output.pci-0000_01_00.1.hdmi-stereo-extra3
[Thread 0x7fffa88af050 (LWP 3715) exited]
[New Thread 0x7fffa88af050 (LWP 3718)]
IQ DCR samp_rate: 5e+06
IQ DCR alpha: 2e-07
Changing NB_RX quad rate: 3.072e+07 -> 5e+06
Generating taps for new filter   LO:-5000   HI:5000 TW:2000   Taps: 115
[New Thread 0x7fff61e8f050 (LWP 3719)]
[New Thread 0x7fff6167f050 (LWP 3720)]
[New Thread 0x7fff60e6f050 (LWP 3721)]
[New Thread 0x7fff5bfff050 (LWP 3722)]
[bladeRF source] start: DEBUG: starting source
[New Thread 0x7fff5b7ef050 (LWP 3723)]
[Thread 0x7fff5bfff050 (LWP 3722) exited]
[... 49 lines cut ...]
#1  0x00007ffff5528098 in __volk_32fc_s32fc_x2_rotator_32fc_d ()
   from /usr/local/lib/libvolk.so.2.2.1
#2  0x00007ffff5502d58 in __volk_32fc_s32fc_x2_rotator_32fc ()
   from /usr/local/lib/libvolk.so.2.2.1
#3  0x00007ffff5a846c4 in gr::blocks::rotator_cc_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) () from /usr/local/lib/libgnuradio-blocks.so.3.8.1
#4  0x00007ffff5666e2c in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
   from /usr/local/lib/libgnuradio-runtime.so.3.8.1
#5  0x00007ffff56015fc in gr::block_executor::run_one_iteration() ()
   from /usr/local/lib/libgnuradio-runtime.so.3.8.1
#6  0x00007ffff5678a48 in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, boost::shared_ptr<boost::barrier>, int) ()
   from /usr/local/lib/libgnuradio-runtime.so.3.8.1
#7  0x00007ffff5661b58 in boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke(boost::detail::function::function_buffer&) ()
   from /usr/local/lib/libgnuradio-runtime.so.3.8.1
#8  0x00007ffff568b288 in boost::detail::thread_data<boost::function0<void> >::run() () from /usr/local/lib/libgnuradio-runtime.so.3.8.1
#9  0x00007ffff7921d38 in thread_proxy ()
   from /lib64/libboost_thread.so.1.69.0
#10 0x00007ffff78c9a88 in start_thread () from /lib64/libpthread.so.0
--Type <RET> for more, q to quit, c to continue without paging--
#11 0x00007ffff4c139b8 in clone () from /lib64/libc.so.6

Found a similar error to before...

**Thread 38 "rotator_cc26" received signal SIGSEGV, Segmentation fault.**

Thoughts? The export variable is still in effect, and persists across reboots as well.

argilo, 2022-10-27:
This bug is very likely due to https://github.com/gnuradio/volk/issues/442.

argilo, 2024-01-01:
This should now be fixed in >= VOLK 3.1.0 and GNU Radio >= 3.10.9.1.