gqrx-sdr/gqrx #783: WFM Stereo separation disappeared?

es5nhc, 2020-04-16:
It appears that the commit 1bf1e1c2f97c635744070eca49db9bf1661062f6 has broken WFM Stereo decoding(the version of Stereo that is transmitted in the 87.5-108 MHZ band, not the OIRT one). The separation between left and right channels has become virtually zero - you might as well just stick to Mono mode. I've verified this by using test signals. In the previous commit - with a commit ID 9718d60c003b31a328b4457b400308c603e1518e, the situation with stereo audio was still normal. It seems that changing to the "new style taps" has broken something. 

As for the version of the GQRX that this is happening at, the situation is still occurring as of the current master branch state(as of 11 UTC on 16 April)

I am aware this isn't really much information, so if you have any further questions, please feel free to ask and I'll provide additional information as needed.

argilo, 2020-04-17:
Thanks for the bug report. I was able to reproduce the problem, and reverting 1bf1e1c2f97c635744070eca49db9bf1661062f6 does bring back stereo separation. I'll dig further to see if I can track down the root cause.

argilo, 2020-04-17:
I think the problem is that de-emphasis is applied too early in the receive chain. Currently it occurs right after quadrature demodulation in `rx_demod_fm`, but it should be applied separately to the left & right channels in `stereo_demod`.

argilo, 2020-04-17:
I'll take a crack at fixing this over the weekend.

argilo, 2020-04-18:
@es5nhc I've proposed a fix for this in #784.

PR #784 (2020-04-18): Apply de-emphasis after stereo demodulation
Fixes #783.

The fix to FM de-emphasis taps in #733 made another bug apparent: the WFM stereo receiver applies de-emphasis to the entire baseband signal. This causes the L-R signal to be greatly attenuated, removing stereo separation.

To fix this, I've moved de-emphasis to the end of the WFM receive chain so that it is applied do the demodulated left & right channels. After this change, de-emphasis is needed in two places (rx_demod_fm.cpp for Narrow FM, and stereo_demod.cpp for WFM), so I moved it into its own hierarchical block.
closed 2020-04-18 by csete