gqrx-sdr/gqrx #789: Device connection in a separate thread

MarcinWieczorek, 2020-04-28:
Hello,

First of all thank you for creating this project.
I have an SDR device and it works well connected to the usb port. I decided to run it off-site and transmit the data over the network.
My issue can be divided into 3 parts:
- gqrx seems to be trying to connect to the tcp socket in the main thread hanging the gui. I often need to restart gqrx when the connection fails.
- Also, in such case settings are often broken and revert to default values (bookmark view hides etc).
- The "play" button that starts the device cannot work as "stop/cancel" once gqrx is trying to connect to the device (and I realize I entered wrong ip/port). It works well as stop button for already connected device.

The obvious solution seems to be running the connection in a separate thread (or separate gui from everything else), but I don't know the construction of gqrx so I'm not saying you have to do it that way.

argilo, 2020-11-28:
I suspect the problem you're encountering is due to a bug in gr-osmosdr which was fixed here:

https://git.osmocom.org/gr-osmosdr/commit/?id=137c568c6094433ce7f038cdfd4f25106c2b4b21

I'll close off the issue for now, but please let me know if you're still seeing this problem with the latest version of gr-osmosdr.

closed 2020-11-28 by argilo

MarcinWieczorek, 2020-11-28:
Thanks. I'll try to confirm that.

MarcinWieczorek, 2020-11-28:
At a first glance gqrx seems to be acting the same. It hangs, defaults the options, play button hangs. I will investigate further, but I prompt you to reopen the issue. I want to let you know that I'm active and willing to provide testing equipment.

argilo, 2020-11-29:
What version of gr-osmosdr are you using?

MarcinWieczorek, 2020-11-29:
git `1:0.6.0.r28.g0847e93`

argilo, 2020-11-29:
That doesn't make sense. The latest version of gr-osmosdr is 0.2.2.

MarcinWieczorek, 2020-11-29:
You're right. But I don't think I understand the connection between gr-osmosdr and rtl-sdr which I am actually running. I mentioned it just yesterday, I thought rtl-sdr depends on gr-osmosdr. I'm serving my hardware over the network to gqrx with just `rtl_tcp -a 0.0.0.0`.

argilo, 2020-11-29:
Gqrx uses gr-osmosdr to make the outbound connection to your rtl_tcp server. It's gr-osmosdr that had a bug which caused it to hang when it couldn't connect.

MarcinWieczorek, 2020-11-29:
I understand now. I'm having trouble building latest commit of gr-osmosdr, I'll keep you updated.