gqrx-sdr/gqrx #791: VOLK usage leads to unusable decoding of everything except simple audio

dvdesolve, 2020-05-04:
More details are [here](https://ham.stackexchange.com/questions/16488/problem-with-rtl-sdr-v3-dongle-unusable-pll/).

Last year I've been using GQRX + Audacity + WXtoImg combination just fine. Then this winter I generated optimal VOLK profile as `README.md` recommends with the `volk_profile` command. After that I wasn't able to decode APT and DRM transmissions at all - the whole situation just looked like troubles with PLL locking (I can't upload corresponding images right now because of poor network connection in the forest so please check a link at the very beginning). APT images were heavily skewed and FFT analysis haven't shown prominent 2.4 kHz peak despite original reception was at very good SNR (about 30 dB). DRM transmissions never were fine - CRCs just were wrong so no decoding was possible.

But as soon as I've removed the whole `.volk` directory from my home folder things returned back - APT decoding started to work again and I'm able to receive and decode DRM transmissions right now!

My software:

- Arch Linux 64 bit
- Kernel 5.6.8
- GQRX 2.12.1
- gnuradio 3.8.0.0
- Audacity 2.3.3
- DReaM 2.1.1

My hardware:

- RTL SDR v3 dongle from rtl-sdr.com with TXCO (-1.6 ppm calibrated)
- Asus VivoMini VC65
- Intel(R) Core(TM) i3-6100T CPU @ 3.20GHz

argilo, 2020-05-04:
Which version of VOLK are you using? Both 2.0.0 and 2.2.0 had serious DSP bugs in the Rotator kernel; those bugs were fixed in the 2.1.0 and 2.2.1 releases respectively.

dvdesolve, 2020-05-04:
VOLK version is 2.2.1, but profile was generated with older version (can't say exact one). Don't know if this matters

argilo, 2020-05-04:
> profile was generated with older version (can't say exact one). Don't know if this matters

I doubt that matters.

Are you able to reproduce the problem again if you run `volk_profile`? If so, please attach your `~/.volk/volk_config` file. Also, please try deleting the `volk_32fc_s32fc_x2_rotator_32fc` line from the file to see if that is the culprit. If not, you could try deleting other lines from that file to determine which kernel might be the problematic one.

dvdesolve, 2020-05-04:
I'll try it as soon as nearest DRM transmission will begin because NOAA/Meteor flights are rare and I'll be able to test more during DRM receiving session

dvdesolve, 2020-05-05:
Seems like I was wrong.

Today I tried to record NOAA 18 morning pass with the following settings:
- Freq. correction = -1.6 ppm
- GQRX FFT size = 16384
- VOLK disabled
and again I got screwed decoding! For the next morning pass I set parameters to the following:
- Freq. correction = 0 ppm
- GQRX FFT size = 8192
- VOLK enabled

Here is log of `volk_profile` run and `~/.volk/volk_config` contents:

### VOLK profile creation log
```
RUN_VOLK_TESTS: volk_64u_popcntpuppet_64u(131071,1987)
generic completed in 153.322 ms
a_sse4_2 completed in 145.985 ms
Best aligned arch: a_sse4_2
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_64u_popcntpuppet_64u(131071,1987)
generic completed in 152.069 ms
a_sse4_2 completed in 147.315 ms
Best aligned arch: a_sse4_2
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_64u_popcntpuppet_64u(131071,1987)
generic completed in 141.527 ms
a_sse4_2 completed in 148.712 ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_16u_byteswappuppet_16u(131071,1987)
generic completed in 35.1447 ms
u_sse2 completed in 45.2117 ms
a_sse2 completed in 42.5202 ms
u_avx2 completed in 33.1507 ms
a_avx2 completed in 28.175 ms
Best aligned arch: a_avx2
Best unaligned arch: u_avx2
RUN_VOLK_TESTS: volk_32u_byteswappuppet_32u(131071,1987)
generic completed in 73.1545 ms
u_sse2 completed in 113.947 ms
a_sse2 completed in 116.008 ms
u_avx2 completed in 75.3357 ms
a_avx2 completed in 68.6828 ms
Best aligned arch: a_avx2
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_32u_popcntpuppet_32u(131071,1987)
generic completed in 96.886 ms
a_sse4_2 completed in 127.549 ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_64u_byteswappuppet_64u(131071,1987)
generic completed in 164.228 ms
u_sse2 completed in 260.483 ms
a_sse2 completed in 263.126 ms
u_ssse3 completed in 179.564 ms
a_ssse3 completed in 162.769 ms
u_avx2 completed in 174.059 ms
a_avx2 completed in 157.365 ms
Best aligned arch: a_avx2
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_32fc_s32fc_rotatorpuppet_32fc(131071,1987)
generic completed in 754.912 ms
a_sse4_1 completed in 440.078 ms
u_sse4_1 completed in 449.001 ms
a_avx completed in 233.286 ms
u_avx completed in 239.159 ms
a_avx_fma completed in 226.191 ms
u_avx_fma completed in 216.835 ms
Best aligned arch: u_avx_fma
[... 1005 lines cut ...]
volk_64f_x2_add_64f u_avx u_avx
volk_8ic_deinterleave_16i_x2 generic generic
volk_8ic_s32f_deinterleave_32f_x2 generic generic
volk_8ic_deinterleave_real_16i a_avx2 u_avx2
volk_8ic_s32f_deinterleave_real_32f a_avx2 u_avx2
volk_8ic_deinterleave_real_8i u_avx2 u_avx2
volk_8ic_x2_multiply_conjugate_16ic u_avx2 u_avx2
volk_8ic_x2_s32f_multiply_conjugate_32fc a_avx2 u_avx2
volk_8i_convert_16i a_avx2 generic
volk_8i_s32f_convert_32f a_avx2 u_avx2
volk_32fc_s32fc_multiply_32fc a_avx u_avx
volk_32f_s32f_multiply_32f a_avx u_orc
volk_32f_binary_slicer_32i generic_branchless generic_branchless
volk_32f_binary_slicer_8i a_avx2 generic_branchless
volk_32u_reverse_32u bintree_permute_bottom_up bintree_permute_bottom_up
volk_32f_tanh_32f a_avx u_avx
volk_32f_s32f_mod_rangepuppet_32f a_avx u_avx
volk_32fc_x2_s32fc_multiply_conjugate_add_32fc u_avx u_avx
volk_8u_x3_encodepolar_8u_x2 a_avx2 u_avx2
volk_32f_8u_polarbutterfly_32f u_avx2 u_avx2
```

Also I'm attaching screenshot of SpectrumLab FFT analysis of recorded passes. The bad one is on the bottom and the good one on the top:
[image]
Now it seems like VOLK is not the cause but it is freq. correction. I doubt that GQRX FFT size matters in this situation.

dvdesolve, 2020-05-07:
I've tried with VOLK and FFT size of 16384 and without frequency correction - all works fine. Now I'm sure that this is frequency correction feature who breaks decoding

csete, 2020-06-06:
Gqrx doesn't do anything with the frequency correction besides sending it to the gr-osmosdr backend. I can imagine that some drivers will also adjust the sampling frequency; however, it is very surprising that a 1.6 ppm correction leads to decoding failure. Perhaps other sample rates wil lwork better.

Anyway, I don't think this is a bug in gqrx.

dvdesolve, 2020-06-07:
using frequency correction in soapysdr or librtlsdr apps (glrpt and mlrpt)
doesn't lead to such problems. may be because these apps support only
integer freq. correction factors and I'm forced to use value of 2 instead
of 1.6

73, Viktor

closed 2020-06-14 by dvdesolve