gqrx-sdr/gqrx #793: Make Tuning frequency by mouse !

khanfar, 2020-05-04:
Hello , I have a big issue when tuning in GQRX ! I cant move the spectrum by scrolling using mouse or click in mouse in spectrum then move right or left ! This make the tuning so hard with just click on number of frequency! Please make the tuning easy and work with mouse like any sdr software ! This is big Issue !!!

HokyP, 2020-06-02:
Agree - Receiver offset should only be top right box not main area..

ulrichloose, 2020-11-30:
I don't think that's a good idea ...

The division into hardware frequency, offset frequency, receive frequency - if those are the correct expressions - makes perfect sense. It is a kind of coarse-fine adjustment in which one sets the center frequency of the SDR hardware once "roughly" and "fine" sets the actually desired reception frequency within the supplied spectrum. Every hardware frequency change generates - at least with my RTL stick - interference and the offset frequency setting is precisely the point of this reception technology.

However, I also have a few wishes (;-)

When recalling a bookmark, the offset frequency should be automatically set to zero and the bookmark frequency should be displayed as the center frequency / hardware frequency / reception frequency in the middle of the spectrum. There is also a kind of error: is the offset frequency e.g. at 500 KHz, no bookmark frequency lower than 500 KHz can be called up. With very broadband SDR hardware, this non-area is even larger.

As a second wish in the matter ...

With a right-click on the offset frequency, this should not simply be set to zero, but should be included in the center frequency. As an example, the current situation:

Hardware frequency 1 MHz, offset frequency 600 KHz, receiving frequency 1.6 MHz. Right click on offset frequency -> receive frequency = 1 MHz, offset frequency = 0, hardware frequency = 1 MHz.

from it should

Hardware frequency 1.6 MHz, offset frequency 0 KHz, receiving frequency 1.6 MHz.