gqrx-sdr/gqrx #794: Occasional hang with RTL SDR ("Waterfall spaghetti")

dvdesolve, 2020-05-04:
I own RTL SDR v3 dongle and sometimes I'm facing this weird situation:

[image]

Waterfall display turns into spaghetti mode, FFT freezes and no reception is possible. The only way to deal with it is to quit from GQRX and run it again (sometimes it's required to do several times).

I use latest stable GQRX and Linux kernel (Arch Linux 64 bit) and observe this problem since the end of 2018 (my first tests with RTL SDR). I've tried with two different dongles: the current v3 and famous "blue Chinese RTLSDR" - the result is the same.

csete, 2020-05-04:
I think this happens if the device stops sending samples. If you start gqrx from a terminal there might be some messages about it. Since you experience it with two different dongles, it could be the computers USB port or the USB driver.

LongnoseRob, 2020-05-05:
Try conencting the USB-dongles to a USB-hub with external power supply.

dvdesolve, 2020-05-05:
Ok, thank you for the suggestions. I'll try some and if it will be possible to catch  a bug I'll post the results.

As for now I think it could be closed.

closed 2020-05-05 by dvdesolve