gqrx-sdr/gqrx #795: Build error  (macOS 10.15 Catalina)

ah1102, 2020-05-08:
I tried to build, but it gives an error at the end:

```
MACOS Installation from source
 ------------------------
 brew install gr-osmosdr
 brew install gnuradio
 brew install cmake
 brew install qt

$ git clone https://github.com/csete/gqrx.git gqrx.git
$ cd gqrx.git
$ mkdir build
$ cd build
$ cmake ..
$ make
```

error:

```
[ 90%] Building CXX object src/CMakeFiles/gqrx.dir/receivers/nbrx.cpp.o
[ 92%] Building CXX object src/CMakeFiles/gqrx.dir/receivers/receiver_base.cpp.o
[ 93%] Building CXX object src/CMakeFiles/gqrx.dir/receivers/wfmrx.cpp.o
[ 94%] Building CXX object src/CMakeFiles/gqrx.dir/portaudio/device_list.cpp.o
[ 96%] Building CXX object src/CMakeFiles/gqrx.dir/portaudio/portaudio_sink.cpp.o
[ 97%] Building CXX object src/CMakeFiles/gqrx.dir/qrc_icons.cpp.o
[ 98%] Building CXX object src/CMakeFiles/gqrx.dir/qrc_textfiles.cpp.o
[100%] Linking CXX executable gqrx
Undefined symbols for architecture x86_64:
  "_volk_32f_log2_32f", referenced from:
      MainWindow::iqFftTimeout() in mainwindow.cpp.o
  "_volk_32f_s32f_multiply_32f", referenced from:
      MainWindow::iqFftTimeout() in mainwindow.cpp.o
  "_volk_32fc_magnitude_squared_32f", referenced from:
      MainWindow::iqFftTimeout() in mainwindow.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/gqrx] Error 1
make[1]: *** [src/CMakeFiles/gqrx.dir/all] Error 2
make: *** [all] Error 2
Apple@iMac-Apple build % 
```

**How to fix it?**

csete, 2020-05-09:
We probably forgot to add libvolk as an explicit dependency in the CMake file.

clayton-shopify, 2020-05-09:
@ah1102 I followed your steps on a machine running Catalina and couldn't reproduce the problem. What does `brew info gnuradio` show?

argilo, 2020-05-09:
@ah1102 I followed your steps on a machine running Catalina and couldn't reproduce the problem. What does `brew info gnuradio` show?

ah1102, 2020-05-09:
```
Apple@iMac-Apple ~ % brew info gnuradio
gnuradio: stable 3.7.13.4 (bottled), HEAD
SDK providing the signal processing runtime and processing blocks
https://gnuradio.org/
/usr/local/Cellar/gnuradio/3.7.13.4_14 (10,775 files, 320.5MB) *
  Poured from bottle on 2020-03-07 at 17:40:46
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/gnuradio.rb
==> Dependencies
Build: cmake ✘, doxygen ✘, pkg-config ✔, sphinx-doc ✔, swig ✘
Required: boost ✔, fftw ✔, gsl ✔, numpy@1.16 ✔, portaudio ✔, uhd ✔, zeromq ✔
==> Requirements
Required: macOS is required ✔
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 460 (30 days), 1,486 (90 days), 8,509 (365 days)
install-on-request: 408 (30 days), 1,293 (90 days), 7,275 (365 days)
build-error: 0 (30 days)
Apple@iMac-Apple ~ % 

```

ah1102, 2020-05-09:
I tried uninstalling and installing gnuradio again and see that there is an error:

```
==> Downloading https://homebrew.bintray.com/bottles/gnuradio-3.7.13.4_14.catali
Already downloaded: /Users/Apple/Library/Caches/Homebrew/downloads/35878d05101258d76a2cb2b2314651187da8cc793278eecadbd2225b560b867a--gnuradio-3.7.13.4_14.catalina.bottle.tar.gz
==> Pouring gnuradio-3.7.13.4_14.catalina.bottle.tar.gz
Warning: gnuradio dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink bin/volk_profile
Target /usr/local/bin/volk_profile
already exists. You may want to remove it:
  rm '/usr/local/bin/volk_profile'

To force the link and overwrite all conflicting files:
  brew link --overwrite gnuradio

To list all files that would be deleted:
  brew link --overwrite --dry-run gnuradio

Possible conflicting files are:
/usr/local/bin/volk_profile -> /Applications/Gqrx.app/Contents/MacOS/volk_profile
==> Summary
🍺  /usr/local/Cellar/gnuradio/3.7.13.4_14: 10,775 files, 320.5MB
Apple@iMac-Apple ~ % 
```

argilo, 2020-05-09:
Could you check whether this change fixes it? https://github.com/csete/gqrx/compare/master...argilo:link-libvolk

To do so, go back to your build folder and run the following:
```
git remote add argilo https://github.com/argilo/gqrx.git
git fetch argilo
git checkout link-libvolk
cmake ..
make
```

ah1102, 2020-05-09:
Thank you, everyone gathered without errors) Please add these changes to the main repository.

[image]

ra1nb0w, 2020-05-09:
you could also use `sudo port install gqrx` or `sudo port install gqrx-devel` with macports. Then you will have the App in `/Applications/MacPorts/`. Soon GNURadio 3.8

argilo, 2020-05-09:
I've opened a pull request: #796

PR #796 (2020-05-10): Add libvolk as an exilicit dependency
Fixes #795.

Gqrx started using libvolk directly in #770, so I'm adding it to `target_link_libraries` to ensure that it's linked. I've verified Gqrx builds successfully against GNU Radio 3.8 (installed with PyBOMBS on Ubuntu 20.04), and GNU Radio 3.7 (installed with Homebrew on macOS 10.15).
closed 2020-05-10 by csete