gqrx-sdr/gqrx #804: Control Panes won't go back into main window

Zotter, 2020-06-04:
Fedora 32, updated - installed GQRX from Fedora repository. 

GQRX runs and performs well. But in the UI, if I happen to 'separate' one of the side panes, such as 'Audio', 'FFT Settings' or 'Bookmarks' they'll separate out from the main window just fine - but I can't seem to find a way to get them to re-attach back INTO the main window. See screen shot for separated panes.

[image]

csete, 2020-06-06:
Please use the mailing list for user support: https://gqrx.dk/user-support

closed 2020-06-06 by csete

Abdull, 2024-03-25:
This issue's topic was continued in [a *Google Groups* conversation](https://groups.google.com/g/gqrx/c/8unIXq0vHXE/m/DladqDKZBAAJ).

The issue still exists with the current gqrx version 2.17.4, and I guess more and more users are affected now in 2024 than were in 2020, when *this* issue was created.

The workaround is:

In file `.config/gqrx/default.conf`, under the `[gui]` section, remove the `geometry=@ByteArray(...)` and `state=@ByteArray(...)` entries.

argilo, 2024-03-25:
Unfortunately I can't reproduce this issue on my Ubuntu system, and I'm not sure what the cause of the issue might be. I hope someone else can help diagnose this.

geerlingguy, 2025-07-16:
@argilo - Raspberry Pi OS, with GRQX installed via the Pi OS 12 (Debian 12) repos. Maybe an older version that's not fixed?

argilo, 2025-07-31:
I was able to reproduce the issue on Raspberry Pi OS, both with the gqrx-sdr package, and with Gqrx 2.17.7 built from source against Qt6.

I suspect the cause may be this Qt bug: https://bugreports.qt.io/browse/QTBUG-87332

That bug was fixed in Qt 6.5.4, so I would hope that things will work properly when Raspberry Pi OS updates to Debian 13 (which will have Qt 6.8.2).

argilo, 2025-07-31:
Switching from Wayland to X in raspi-config fixes the problem, which does suggest it's a Wayland-related issue.

Regardless of whether this is a Qt bug, it might perhaps be useful to have a "Reset" option to put the UI back to its default state.

argilo, 2025-08-01:
Coincidentally, I was in the middle of updating to Trixie following the instructions in that thread. I'll take Gqrx for a spin once it finishes.

argilo, 2025-08-01:
I just finished the update, and in Trixie it's possible to re-dock the panels by clicking in the upper right corner. But unfortunately it seems that panels can't be dragged while in the un-docked state. ☹️

argilo, 2025-08-01:
It appears the problem with dragging is because the "toplevel drag" protocol is not yet implemented in Labwc (nor in Wayfire): https://wayland.app/protocols/xdg-toplevel-drag-v1

argilo, 2025-08-01:
Forcing Gqrx to use XWayland by launching it with `QT_QPA_PLATFORM=xcb gqrx` works around the problem.

argilo, 2025-08-01:
Some discussion about supporting toplevel drag in Labwc: https://github.com/labwc/labwc-scope/issues/12