gqrx-sdr/gqrx #805: APT signal oddities after GNU Radio update

dvdesolve, 2020-06-06:
After latest update of GNU Radio infrastructure I've faced strange problem in GQRX. I use it for receiving and recording APT transmissions from NOAA 18 and 19. Here is full list of updates which came on my Arch system (all packages were built with GCC 10.1.0):

- `libvolk` (2.2.1 -> 2.3.0)
- `gnuradio` (3.8.0.0 -> 3.8.1.0)
- `gnuradio-companion` (3.8.0.0 -> 3.8.1.0)
- `gnuradio-iqbal` (0.37.2 -> 0.38.1)
- `gnuradio-fcdproplus` (rebuild with GCC 10.1.0)
- `gnuradio-osmosdr` (rebuild with GCC 10.1.0)
- `gqrx` (rebuild with GCC 10.1.0)

Before the update APT waterfall image were like a group of thin lines without duplicates and I was able to hear clear tone of 2.4 kHz. Example image:

[image]

Now visual pattern has changed - instead of thin single lines I see changing pattern with two lines instead of one in every group changing distance from each other with time. When I change frequency they converges or diverges (you can see these "steps" on waterfall image below). Also instead of single 2.4 kHz audio tone I hear distorted audio. Example image:

[image]

I've tried to decode resulting image (after processing with SoX) and got the result but it's corrupted a bit: you can see small horizontal stripes (don't mess it with noise lines!) where they shouldn't be. Reception parameters in GQRX: WFM (mono), 38 kHz bandwidth, 32 decimation, 2.56 Msps sampling rate, 38.4 dB gain. Example image:

[image]

I've already tried:
- record NOAA 18 during morning pass with old VOLK profile (which was fine)
- record NOAA 18 during morning pass without VOLK profile
- record NOAA 19 during evening pass with new VOLK profile

Seems like these updates have broken something because I use the same RTL-SDR dongle, V-dipole antenna and other hardware. What can cause this behavior? Could it be related to the way how GQRX communicates with GNU Radio? At the same time I'm able to receive LRPT images from Meteor-M2 with `glrpt` just fine and without any issues.

dvdesolve, 2020-06-08:
I've re-checked yesterday with cheap blue dongle instead of RTL-SDR Blog V3 and both satellites (18 and 19) give the same result. It's unlikely that both sats have failed at the same time.

Moreover I've done FFT for "good" and "bad" recordings. Both pics are below (in mentioned order):
[image]
[image]

Both recordings were renormalized to -1 dB. As you could notice noise level differs dramatically between two recordings.

dvdesolve, 2020-06-11:
I've experimented a bit and found that disabling IQ balancing in GQRX makes waterfall look fine (without extra duplicates and audio tones) however decoding with WXtoIMG results in faulty image (it says that it can't find telemetry). So it looks like the 0.38.1 update of `gnuradio-iqbal` may be broken...

csete, 2020-06-11:
With RTL-SDR dongles that use the R820T/T2 tuner you do not need IQ balancing since they are not direct conversion tuners. On the other hand, if you have automatic IQ balancing enabled when there are carriers on both sides of the spectrum, the algorithm might mistankingly try to eliminate those that appear to be images.

dvdesolve, 2020-06-11:
thank you for the hint. it worked fine with iq balancing until recent
update. however, I've tried to disable it and ghost lines disappeared but
apt decoding wasn't possible at all

73, Viktor

dvdesolve, 2020-06-14:
Removing VOLK profile config and disabling IQ balancing solves the issue

dvdesolve, 2020-06-14:
It seems like both this issue and #791 leads to improper behavior. I haven't tested DRM reception yet but APT works fine for now. I think both issues could be closed now because most probably there are bugs in underlying components in GNU Radio.

closed 2020-06-14 by dvdesolve