gqrx-sdr/gqrx #810: Pi 4B library issue

Vulchikova, 2020-06-20:
i read through the existimg issues but didnt find anything helpful.

followed the install steps and ran into the following error when trying to start the program:
./gqrx 
./gqrx: error while loading shared libraries: libboost_program_options.so.1.62.0: cannot open shared object file: No such file or directory

my OS info:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

sashko, 2020-07-01:
https://github.com/csete/gqrx/issues/693

argilo, 2020-10-18:
To install Gqrx an a Rasberry Pi 4, simply use the package repository: `sudo apt install gqrx-sdr`.

closed 2020-10-18 by argilo