gqrx-sdr/gqrx #818: Minimal tuning step achievable with mouse clicking is too coarse

ly2ss, 2020-07-27:
Hi,
Using Gqrx v2.12.1-52-gbcf8531349 with Adalm Pluto for monitoring beacons on 5760 MHz. 
Don't know what causes it but the minimum tuning step is 510 Hz using mouse on waterfall or panadapter. One can click anywhere on signal, but vfo "sticks" to nearest 510 Hz points.

csete, 2020-07-27:
Clicking on the spectrum has a certain "click resolution" which is the sample rate / width of the spectrum in pixels. You can always use the mouse wheel to make finer steps, which depends on the selected mode but definitely less than 510 Hz.

ly2ss, 2020-07-27:
Yes, mouse wheel is working in 10 Hz steps.
But when one see strong CW station and click on the peak of it (the "first shot") and hoping to hear clear tone - vfo itself tunes ("sticks") 510 Hz above or below - this feels very counter-intuitive - almost always missing first shot with consecutive a must fine-tunes with wheel. On 602 Mhz this "sticking step" becomes to 70 Hz on 1602MHz 130Hz, on 2602 250Hz and so on.
I didn't noticed this earlier because all of my observations were below 145 MHz band - where this step is almost unnoticeable (20 Hz on FHD resolution and 1.5Msps).

csete, 2020-07-27:
Okay, what you describe is certainly not intentional as the tuning resolution ("sticking step") should be constant. It could be a bug though I'm surprised this wasn't noticed before.

argilo, 2020-10-18:
I was able to reproduce this as well. At high frequencies (e.g. 5 GHz), there appears to be a significant loss of precision that makes it difficult to tune by clicking. Also, the filter low & high cut frequencies displayed in the plotter appear to suffer from the same loss of precision.

argilo, 2020-10-18:
I think the problem is coming from the single-precision float calculation here:

https://github.com/csete/gqrx/blob/e0a7621caaf637675cd8e1ce7546986c9a611d9a/src/qtgui/plotter.cpp#L1537-L1544

argilo, 2020-10-18:
I changed that calculation to use `qint64` everywhere, and clicking now tunes accurately. But the display is still inaccurate, so there's still more to be fixed. I'll hunt down the other places and then open a pull request.

argilo, 2020-10-18:
The display issues were due to the `xFromFreq` method, which suffered from the same problem. I've fixed both in #838.

PR #838 (2020-10-19): Fix loss of precision in xFromFreq and freqFromX
Fixes #818.

At high frequencies (e.g. 5 GHz) and when zoomed in on the frequency axis, there is a noticeable loss of precision when clicking to tune. Also, the red bar showing the currently tuned frequency and the grey area showing the filter edges exhibit a similar loss of precision.

These problems occur because `CPlotter::xFromFreq` and `CPlotter::freqFromX` use single-precision floating-point, which only has 24 bits of precision. I've switched these calculations to use `qint64` instead.

@ly2ss If you have a chance to test this, it would be [... cut]
closed 2020-10-19 by argilo