gqrx-sdr/gqrx #819: Difficulty compiling on MacOS 10.15 with homebrew qt5

jsr38, 2020-08-07:
I'm almost certain that gqrx is not at fault, rather, my homebrew installed toolchain/Qt5 is to blame, however, googling has not yielded results.

`In file included from /usr/local/opt/qt5/lib/QtGui.framework/Headers/QColor:1:
In file included from /usr/local/opt/qt5/lib/QtGui.framework/Headers/qcolor.h:47:
/usr/local/opt/qt5/lib/QtGui.framework/Headers/qrgba64.h:67:14: error: unknown type name 'Q_ALWAYS_INLINE'
    explicit Q_ALWAYS_INLINE Q_DECL_CONSTEXPR QRgba64(quint64 c) : rgba(c) { }
             ^`

I've tried both `cmake` and `qmake` for generating the build system with identical results.  Could someone offer a pointer?

csete, 2020-08-10:
As far as I recall gqrx is also available via homebrrew, so perhaps there are some hints in the homebrew recipe?

jsr38, 2020-08-10:
Thanks for the tip.  That thought had occurred to me and it seems gqrx is now distributed as a cask (binary) although I did look briefly at the legacy formula which didn't hold any obvious clue.

The hint I think I should be taking is from homebrew install output when installing qt5.  There are some messages about cmake builds being broken with the homebrew installed qt5.  I'll look further in this direction.

Obviously, things would be easy if I installed on a Linux system but I can't access hardware right now.   

I'm pretty sure I need to install from source as I'm trying to get a LimeSDR Mini working on MacOS.  I may well be wrong about that!

theresiasnow, 2020-08-16:
Got the latest source compiling on my mac

MCKLtech, 2020-11-28:
@theresiasnow could you help me in the build process for MacOS? See https://github.com/csete/gqrx/issues/876

argilo, 2020-11-28:
@jsr38 Are you still seeing this problem? CI builds are working fine on macOS 10.15 these days. I'll close this issue off for now, but please let me know if you're still seeing the error.

closed 2020-11-28 by argilo