gqrx-sdr/gqrx #820: Doesn't build on Linux with gcc 10.2

hofman2005, 2020-08-10:
FAILED: src/CMakeFiles/gqrx.dir/dsp/rds/parser_impl.cc.o
/usr/bin/x86_64-pc-linux-gnu-g++ -DCUSTOM_AIRSPY_KERNELS -DGENTOO -DGNURADIO_VERSION=198657 -DLOG4CPP_FIX_ERROR_COLLISION=1 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_
LIB -DVERSION=\"v2.12.1-52-gbcf8531-dirty\" -DWITH_PULSEAUDIO -Isrc -I/var/tmp/portage/net-wireless/gqrx-9999/work/gqrx-9999/src -Isrc/gqrx_autogen/include -I/var/tmp/portage/net-wireless/gqrx-9999/work/gqrx-
9999/include -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtCore -isystem /usr/lib64/qt5/mkspecs/linux-g++ -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/
qt5/QtGui -isystem /usr/include/qt5/QtSvg -march=native -O2 -pipe -D_GLIBCXX_USE_CXX11_ABI=0 -DNDEBUG -march=native -O2 -pipe   -Wall -Wextra -Wno-unused-parameter -Wsign-compare -fPIC -std=gnu++11 -MD -MT sr
c/CMakeFiles/gqrx.dir/dsp/rds/parser_impl.cc.o -MF src/CMakeFiles/gqrx.dir/dsp/rds/parser_impl.cc.o.d -o src/CMakeFiles/gqrx.dir/dsp/rds/parser_impl.cc.o -c /var/tmp/portage/net-wireless/gqrx-9999/work/gqrx-9
999/src/dsp/rds/parser_impl.cc
/var/tmp/portage/net-wireless/gqrx-9999/work/gqrx-9999/src/dsp/rds/parser_impl.cc: In constructor ‘gr::rds::parser_impl::parser_impl(bool, bool, unsigned char)’:
/var/tmp/portage/net-wireless/gqrx-9999/work/gqrx-9999/src/dsp/rds/parser_impl.cc:43:72: error: ‘_1’ was not declared in this scope
   43 |  set_msg_handler(pmt::mp("in"), boost::bind(&parser_impl::parse, this, _1));

alexf91, 2020-08-10:
I can't reproduce it, but I have some ideas.
Try to add `#include <boost/bind.hpp>` to `src/dsp/rds/parser_impl.cc` and build again.
`parser_impl.cc` doesn't include any boost headers, so it depends on what other headers include. Maybe something was removed.

If you can't modify the source code with portage or emerge before building:
```
git clone https://github.com/csete/gqrx.git
cd gqrx
# modify src/dsp/rds/parser_impl.cc
mkdir build
cd build
cmake ..
make -j
```

hofman2005, 2020-08-12:
Thanks a lot! Adding #include <boost/bind.hpp> works.
I have GNU Radio 3.8.1.0 and Boost 1.73.0

argilo, 2020-10-16:
This should be fixed by #827.

closed 2020-10-16 by argilo