gqrx-sdr/gqrx #826: A little change for compiling on Mac (10.13.6)

ulrichloose, 2020-10-05:
in rx_rds.cpp the line

set_msg_handler(pmt::mp("store"), boost::bind(&rx_rds_store::store, this, _1));

must be changed to

set_msg_handler(pmt::mp("store"), boost::bind(&rx_rds_store::store, this, boost::placeholders::_1));

and in parser_impl.cc the line

set_msg_handler(pmt::mp("in"), boost::bind(&parser_impl::parse, this, _1));

the same

set_msg_handler(pmt::mp("in"), boost::bind(&parser_impl::parse, this, boost::placeholders::_1));

With this changes gqrx compiles without an error.

argilo, 2020-10-05:
There are already two open pull requests for this: #811 and #821.

ulrichloose, 2020-10-05:
Sorry I didn't saw #821 

But why hasn't fix # 821 been incorporated into master after many weeks?

argilo, 2020-10-05:
Like many open source projects, Gqrx is maintained by a volunteer who contributes when time is available.

csete, 2020-10-06:
#811 is now merged.

closed 2020-10-06 by csete

ulrichloose, 2020-10-07:
Anhang verfügbar bis 06.11.2020
Super! Thank you…

I am not sure if there is a need and if I can transfer a large file as an attachment. In any case, the latest version of gqrx as a Mac app with all Dylibs included in the attachment.

regards

Zum Laden klicken
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAekXAwif8w_17pSkj2y-Gqk7GraqAdxKvSNiYZ8kFFO69KFVIOLisp0E%2F%24%7Bf%7D%3Fo%3DAlAIhNA-B126xlBDe0hKb21P4MlA_KyMJc-e0m1RHJG1%26v%3D1%26x%3D3%26a%3DCAogA7vbXK4Jb_BpHgow8ANlzSbeHKgiyqyr6XZpyZstxBASdxD8hfOb0C4Y_JXu79kuIgEAKggByAD_BFMdT1IEOxq2qloE4rKdBGomQJVmqMFDFJGOPSRG1ZlOlMzIrMjnQZTtXUZPl1gziLICprMifB5yJhqV8jUEYJMu-Y6iiG6xBfad2RnNmoGuqyDmGwZQr-Arr0oMKcbJ%26e%3D1604673309%26fl%3D%26r%3D23543253-73A5-4A86-8952-E9E64974E4D1-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3DA772C11E-F22C-4401-81B0-7444503F0B67%26p%3D9%26s%3DAwFT8q-AD65UN_GlHsSTpe6HlWk&uk=cXGyVYUcBNSGzXmCr2SIig&f=Gqrx.app.zip&sz=41308782>Gqrx.app.zip
41,3 MB