gqrx-sdr/gqrx #829: Building with qmake fails without gnuradio-osmosdr, but cmake succeeds.

darrenfreeman, 2020-10-14:
Compiling from the latest git.

It appears that the qmake project requires the gnuradio-osmosdr package, that I don't have installed.

```
qmake ../
Info: creating stash file /opt/gqrx/build/.qmake.stash
Project MESSAGE: No prefix given. Using /usr/local
Project MESSAGE: Gqrx configured with pulseaudio backend.
Project ERROR: gnuradio-osmosdr development package not found
```

However it configures, and then builds fine, if I use cmake instead.

```
cmake ../
[many lines deleted]
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/gqrx/build
```

I don't think this package is really needed if I don't have the relevant hardware. As evidenced by success with cmake.

csete, 2020-10-16:
gnuradio-osmosdr is a required dependency: https://github.com/csete/gqrx/blob/master/CMakeLists.txt#L111
Gqrx can neither build nor run without it.

A more likely explanation is that you actually do have gnuradio-osmosdr installed but qmake can not find it.

darrenfreeman, 2020-10-17:
I created a new build directory, and started again. Here is the relevant output from cmake. Indeed you are right, cmake is able to find my locally built and installed version. (I am using gnuradio etc. all built from source.)

```
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gnuradio-osmosdr'
--   No package 'gnuradio-osmosdr' found
-- Found GNURADIO_OSMOSDR: /usr/local/lib/libgnuradio-osmosdr.so  
```

So is the issue then with libgnuradio-osmosdr not installing a .pc file when it's built from source? Or should qmake be able to find it regardless?

argilo, 2020-11-18:
I attempted to add a a qmake build on MacOS to Travis CI and encountered the same error: https://travis-ci.org/github/argilo/gqrx/jobs/744354506. The qmake build works fine on Linux.

I'll take a look at this soon.

Perhaps this is related? http://osmocom.org/issues/4423

argilo, 2020-11-18:
Yes indeed, the removal of gnuradio-osmosdr.pc in gr-osmosdr 0.2.0 seems to be the problem.

I submitted a patch to gr-osmosdr to bring it back: http://lists.osmocom.org/pipermail/osmocom-sdr/2020-November/002109.html

argilo, 2020-11-18:
Well so much for that approach. gr-osmosdr rejected the patch: http://lists.osmocom.org/pipermail/osmocom-sdr/2020-November/002110.html

argilo, 2020-11-19:
I guess the reason this works on Ubuntu is that Debian patched the pkg-config file back in: https://sources.debian.org/patches/gr-osmosdr/0.2.2-1/restore-pkg-config-support/

argilo, 2020-11-19:
@darrenfreeman Is there any reason why you need qmake? Now that I've looked at the problem more closely, it seems that the GNU Radio ecosystem has standardized on cmake, so continuing with qmake in gqrx will be difficult. It's also a pain to maintain two build systems in parallel. I'm of the mind to remove qmake completely. Qt Creator seems to have official support for cmake, so I don't imagine it will be a problem.

argilo, 2020-11-19:
I'll close this off now, since qmake support was removed in #875.

closed 2020-11-19 by argilo