gqrx-sdr/gqrx #839: menus do not work on macOS

pidloop, 2020-10-19:
Hello!

I have been using 2.11.3 on macOS 10.15.* with my CloudIQ for a few years. It still works very well, many thanks for the great software.

I now tried the new 2.13 via macports. It also works very well except for one thing: none of the menubar items respond to mouse clicks at all, they are completely dead.

Otherwise, a great update.

argilo, 2020-10-19:
Thanks for the bug report! I'll have a look at this soon. In the mean time, if you learn anything more about the bug or track down exactly which release it started in, please let me know.

pidloop, 2020-10-19:
Well, I tried again and the problem disappeared, the menus work fine now. Tried 3rd time, still ok, so it seems you can close this report. Something to come back to bite again some other day ;-)

I do see that "About gqrx" does not report the version so that would be nice to see.

Just FYI, I do see the following in the Console system.log every time I open gqrx:

`
Oct 19 12:01:22 ecd-imac27 com.apple.xpc.launchd[1] (com.apple.xpc.launchd.domain.pid.gqrx.80681): Failed to bootstrap path: path = /opt/local/bin/gqrx, error = 2: No such file or directory
`

even though the file is clearly there and sensible:

`ecd-imac27$ ls -l /opt/local/bin/gqrx`
`-rwxr-xr-x  1 root  wheel  1666748 Oct 18 22:16 /opt/local/bin/gqrx`
`ecd-imac27$ file /opt/local/bin/gqrx`
`/opt/local/bin/gqrx: Mach-O 64-bit executable x86_64`

argilo, 2020-10-28:
I'll close this off, since it doesn't appear the problem was in Gqrx.

closed 2020-10-28 by argilo

pidloop, 2020-10-28:
Sorry I just left this hanging. I've been trying to find a pattern to
report but still don't see one. When it does happen, quitting and
restarting, perhaps a few times, will get them working again eventually.

Sometimes when I start, I get the popup that says the settings are
incorrect. I don't see a pattern to when that happens either (!), but when
it does I always just click ok to accept the current settings anyway, then
the menus almost always work.

But really, it's all very minor. Gqrx is a great way to run my CloudIQ from
macOS and I'm just grateful for your work.

argilo, 2020-10-28:
I guess there might be some problem in Gqrx here then. If you're able to track it down or find a reliable way to reproduce the problem, feel free to re-open this issue.