gqrx-sdr/gqrx #844: Something is wrong with Mac-OS

ulrichloose, 2020-10-25:
When I compile gqrx "out of the box", the CPU load is 140% and just setting a filter from soft to normal leads to sound interference. At first I thought it was due to the Portaudio driver, but the behavior does not change with the Gr-audio setting either. The output "Resampling audio 90000 -> 48000" appeared suspiciously and is always displayed. All audio parameters of the Mac are set to 48000 and no resampling should be necessary. I then found the place in the nbrx.cpp file that contained the output of the Resamplings generated. I'm not a programmer and have little idea, but I just cheekily set the definition #define PREF_QUAD_RATE 48000.f and recompiled. Result: the CPU load sinks to 60% and there are no more sound disturbances. Any other effects I can't tell - everything is running as it should, the CPU load is halved and there are no more sound disturbances.

Is there an explanation? Can one make it like this? (;-)

argilo, 2020-10-25:
Thanks for the report. Can you please add some more information about your setup?

* Which version / commit of Gqrx are you running?
* Which version of GNU Radio are you using and how did you install it?
* What input device are you using?
* What is your input sample rate set to?

Thanks!

ulrichloose, 2020-10-25:
Thank you for your answer....

My Macs are running with OSX 10.13.6

I´m running alway the head of gqrx

GNU-Radio is 3.8.2.0 - locally compiled without Macports or Homebrew

My device is a rtlsdr Stick made by Nooelec

librtlsdr is 0.8 hayguen locally compiled

The inputrate is mostly 1,8 Mhz but never higher than 2,4 Mhz

-----

This behavior is identical on both my Macbook Pro and my Mac Mini. IShowU Audio Capture is also installed on both devices. All data rates that can be set via audio-midi-setup are set to 48 kHz and after my small change in nbrx.cpp the CPU load drops rapidly, but gqrx works completely error-free from my point of view.

The place where I made my change is certainly not really correct, because if something other than 48 Khz is accidentally set in the audio midi setup, an error message appears when starting gqrx.

That is the startup with "my" change (:-), without, there is one more line "Resampling audio 96000 -> 48000"

Ulrichs-MBP:~ ulrichloose$ /Users/ulrichloose/source/gqrx/build/src/gqrx ; exit;
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy airspyhf soapy redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
->>> Resampling audio 96000 -> 48000 <<<--
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 4
  0:  Built-in Microphone  I:2  O:0
  1:  Built-in Output  I:0  O:2
  2:  iShowU Audio Capture  I:2  O:2
  3:  Pass-Thru  I:0  O:2
Using default audio device
BookmarksFile is /Users/ulrichloose/.config/gqrx/bookmarks.csv
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 4
  0:  Built-in Microphone  I:2  O:0
  1:  Built-in Output  I:0  O:2
  2:  iShowU Audio Capture  I:2  O:2
  3:  Pass-Thru  I:0  O:2
[INFO] [UHD] Mac OS; Clang version 10.0.0 (clang-1000.11.45.5); Boost_107200; UHD_3.15.0.0-0-unknown
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy airspyhf soapy redpitaya 
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy airspyhf soapy redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 4
  0:  Built-in Microphone  I:2  O:0
  1:  Built-in Output  I:0  O:2
  2:  iShowU Audio Capture  I:2  O:2
  3:  Pass-Thru  I:0  O:2
Audio device 'Pass-Thru' has index 3
WARNING: Overriding original sample rate of 96000 with 1.8e+06
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 4
  0:  Built-in Microphone  I:2  O:0
  1:  Built-in Output  I:0  O:2
  2:  iShowU Audio Capture  I:2  O:2
  3:  Pass-Thru  I:0  O:2

ulrichloose, 2020-10-25:
In the meantime I have looked through older gqrx sources and my "error" does not seem to be an "error", but was programmed on purpose so that you can receive more narrowband ...

commit 392dd4206a7e745c8ea047f5312664e8dfeaacbf

Increase quadrature rate for narrow band receiver
Reducing the quadrature rate to 48 kHz was probably done while
optimizing for embedded devices; however, 48 kHz is just too narrow if
we want to do things like NOAA APT using the narrow band receiver.

My two computers do not seem to be powerful enough for this. But since I'm not interested in NOAA and I think that it works very well for shortwave CW , SSB and RTTY with 48 Khz, I will change that for me accordingly.

Maybe you can make that a choice? Doubling the processor load for little equivalent on shortwave is a bit "overkill" and unnecessarily even excludes 2 GHz computers.

argilo, 2020-10-26:
Audio sampling should not use much CPU because the sample rate is relatively low. On my system it accounts for about 3% CPU load, and changing `PREF_QUAD_RATE` does not make much difference in the total CPU utilization of Gqrx. Could you check with `top -H` or `htop` to see how much CPU the `pfb_arb_resamp2` thread is taking? What CPU does your machine have? Is VOLK correctly compiled to use vectorized operations?

The only inefficiency I see in the resampler is that it uses a polyphase arbitrary resampler; switching to a rational resampler would likely consume less CPU time.

ulrichloose, 2020-10-26:
Thank you - that was the crucial hint ....

If you don't do everything yourself (;-) volk was installed via homebrew and that doesn't seem to work properly. I have now removed that again and compiled it myself. The volk_config file now has over a hundred entries with all SSE functions. In the homebrew version there were hardly any entries and those that were there were all set to generic.

Nice crap! So I am very confident that this will solve my problem. The only question that remains is whether I have to recompile gnu-radio & Co. with the new working Volk version.

Thank you again for the crucial hint - I would not have come across that in years.

ulrichloose, 2020-10-26:
Perfect!

The first thing I did was to recompile volk from the sources and that was the "breakthrough" ... For security reasons I also recompiled gnu-radio, gr-osmosdr and gqrx.

Now everything looks right: no more 140% load, everything is OK ...

Thanks again for the help!

closed 2020-10-26 by ulrichloose