gqrx-sdr/gqrx #845: GQRX doesn't compile with forked gr-osmosdr (PlutoSDR/Spyserver)

BX2ABT, 2020-10-26:
To get support for PlutoSDR and Spyserver I compiled and installed a fork of gr-osmosdr found here: [https://github.com/dk2ro/gr-osmosdr-pluto-spyserver](). System is Kubuntu 20.04LTS. Dependencies are installed from the Ubuntu repos.

I got the latest GQRX source via git clone and everything compiles untill 100%, but then the following:

`[100%] Linking CXX executable gqrx
/usr/bin/ld: /lib/x86_64-linux-gnu/libgnuradio-iio.so.1: undefined reference to `boost::chrono::steady_clock::now()'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/gqrx.dir/build.make:1071: src/gqrx] Error 1
make[1]: *** [CMakeFiles/Makefile2:248: src/CMakeFiles/gqrx.dir/all] Error 2
make: *** [Makefile:130: all] Error 2`

libboost-chrono-dev v1.7 is installed from the Ubuntu repos. What did I miss or mess up? Thanks for the help. I really hope to get PlutoSDR and Spyserver working with GQRX.

argilo, 2020-10-27:
I suspect this is a bug in gr-iio. It may have been fixed in https://github.com/analogdevicesinc/gr-iio/pull/54/commits/56e51f378968c777184cad1a89375b39c66cf59a. Could you try building the latest gr-iio from source to see if that solves the problem?

vladisslav2011, 2021-12-14:
I've managed to get gqrx working with this gr-osmosdr fork, but... This fork is buggy as hell.
1. I had to fix compilation error (ubuntu 18.04)
2. Change buffer sizes to working values
3. Fix math error in type conversion (missing parenthesis)
4. Fix thread stack abuse
5. Work around instant deadlock
6. Work around buffer skipping problem (the ring buffer is used in a way, it should never work correctly)
Now I can listen to Europe broadcast FM stations, but still sometimes get freezes as I did not rewrite tcp client to use nonblocking IO...
I did not have installed this library as system default, but load it with LD_PRELOAD instead. Gqrx is compiled and linked against default gr-osmosdr, installed with apt. Maybe this would help.

I have not tested it with PlutoSDR as I do not own one...

argilo, 2022-01-25:
I'll close off this issue, as the problem seems to be in the fork and not Gqrx.

closed 2022-01-25 by argilo