gqrx-sdr/gqrx #846: Clicking on audio waterfall causes crash

lsquared618, 2020-10-27:
Hi developers of gqrx, really great software and performance seems to have improved a lot recently 👍🏼 

Below is a problem I found when running the newest version of gqrx:

I did a `git clone` then build and found that whenever I click on the audio waterfall, GQRX crashes immediately. Clicking on the audio spectrum (pandapter) is fine. When gqrx crashes I get the error message `“gqrx” terminated by signal SIGFPE (Floating point exception)` in the terminal. 

I have deleted all config files of gqrx but the problem persists.

argilo, 2020-10-27:
Thanks for the bug report!

Unfortunately I haven't been able to reproduce the problem: clicking on the waterfall is working for me at the moment. Could you run gqrx inside gdb and provide a backtrace? If you need instructions, let me know.

argilo, 2020-10-27:
Also, could you paste in a copy of your `~/.config/gqrx/default.conf` file, and let me know any steps you performed between starting gqrx and clicking on the waterfall? Thanks!

argilo, 2020-10-27:
Lastly, please let me know what operating system you're using, whether you installed GNU Radio and gr-osmosdr from the package manager or built from source, and what versions of them you're running.

lsquared618, 2020-10-27:
Steps to reproduce:
- Run gqrx
- Click the "..." button in the Audio tab
- Slide the "Pandapter - Waterfall" slider **to the left end**
  - I forgot to mention, but this bug only happens when this slider is set to the left end
- Close the audio settings window
- Click on the audio waterfall
- gqrx crashes

lsquared618, 2020-10-27:
Backtrace:
```
Thread 1 "gqrx" received signal SIGFPE, Arithmetic exception.
0x00005555556a5081 in CPlotter::freqFromX(int) ()
(gdb) bt
#0  0x00005555556a5081 in CPlotter::freqFromX(int) ()
#1  0x00005555556aa192 in CPlotter::mousePressEvent(QMouseEvent*) ()
#2  0x00007ffff7312b0e in QWidget::event(QEvent*) () from /usr/lib/libQt5Widgets.so.5
#3  0x00007ffff73bdf3f in QFrame::event(QEvent*) () from /usr/lib/libQt5Widgets.so.5
#4  0x00007ffff72d1752 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#5  0x00007ffff72d887b in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#6  0x00007ffff680bcda in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#7  0x00007ffff72d787e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /usr/lib/libQt5Widgets.so.5
#8  0x00007ffff732b249 in ?? () from /usr/lib/libQt5Widgets.so.5
#9  0x00007ffff732e63f in ?? () from /usr/lib/libQt5Widgets.so.5
#10 0x00007ffff72d1752 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#11 0x00007ffff680bcda in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#12 0x00007ffff6be44ac in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/lib/libQt5Gui.so.5
#13 0x00007ffff6bb9bac in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Gui.so.5
#14 0x00007ffff04caf9c in ?? () from /usr/lib/libQt5XcbQpa.so.5
#15 0x00007ffff1818914 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#16 0x00007ffff186c7d1 in ?? () from /usr/lib/libglib-2.0.so.0
#17 0x00007ffff1817121 in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
#18 0x00007ffff6864941 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#19 0x00007ffff680a65c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#20 0x00007ffff6812af4 in QCoreApplication::exec() () from /usr/lib/libQt5Core.so.5
#21 0x00005555555ed483 in main ()
```

lsquared618, 2020-10-27:
`~/.config/gqrx/default.conf`:
```
[General]
crashed=true

[input]
device="rtl=0"
sample_rate=1800000
```

lsquared618, 2020-10-27:
Operating system: Arch Linux, `Linux shuyuan-zenbook 5.9.1-arch1-1 #1 SMP PREEMPT Sat, 17 Oct 2020 13:30:37 +0000 x86_64 GNU/Linux`

All packages up to date, GNU Radio and gr-osmosdr installed with package manager.

- gnuradio 3.8.1.0-4
- gnuradio-osmosdr 0.2.0-4

argilo, 2020-10-27:
> Slide the "Pandapter - Waterfall" slider to the left end

This seems to be the key. I was able to reproduce the problem once I did that. I'll have a look at this soon.

argilo, 2020-10-27:
A crash also occurs if the RF waterfall is expanded to the maximum size in the "FFT Settings" panel.

argilo, 2020-10-27:
@lsquared618 I opened a pull request that should fix this: #848. If you have a chance to test it, that would be much appreciated.

PR #848 (2020-10-28): Fix division by zero in width calculations
Fixes #846.

If the Panadapter-Waterfall slider (in the "FFT Settings" panel or the audio FFT options dialog) is set such that the waterfall occupies 100% of the display area, then crashes occur due to division by zero in several places where the width of the panadapter is used. (In these cases, the panadapter is null, so `m_OverlayPixmap.width()` returns zero.)

Crashes occur in the following situations:

* hovering over the RF waterfall
* clicking on the audio waterfall
* adjusting the "Freq zoom" slider

To fix this problem, I've changed these calc [... cut]
closed 2020-10-28 by argilo