gqrx-sdr/gqrx #847: On ubuntu 20.04.1 LTS x86_64: no signal, blank waterfall

nmaster2042, 2020-10-27:
Hi.

I get a strange issue with gqrx on my sustem.
First with official repo version 2.12 and the last 2.13.2 AppImage.

See below
[image]

I also tried build last version from source and same issue.

nmaster2042, 2020-10-27:
On the terminal I only see one error but I don't know if it's related to this issue:

```
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd sdrplay hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Resampling audio 96000 -> 48000
BookmarksFile is /home/nicolas/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-16-ga3ece4f2
get_devices started
Device count: 0
get_devices end
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 790, resource id: 41943357, major code: 40 (TranslateCoords), minor code: 0
```

argilo, 2020-10-27:
Did you select your RTL-SDR as the input in the "Configure I/Q devices" dialog? It should appear as something like "Realtek RTL2838UHIDIR SN:00000001" in the list.

nmaster2042, 2020-10-28:
Yes. The RTL-SDR is detected and selected.
[image]

argilo, 2020-10-28:
Unfortunately I don't know what might cause this. If you learn anything more about the problem, please post an update and I'll have a look.

argilo, 2020-10-28:
For comparison, here's what I see when I start up Gqrx:
```
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd sdrplay hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
BookmarksFile is /home/argilo/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-16-ga3ece4f2
get_devices started
Device count: 0
get_devices end
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd sdrplay hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
```
I do not see the `XCB error`, so that could be relevant.

Are you able to use other software with your RTL-SDR?

nmaster2042, 2020-10-29:
Yes I can use my rtlsdr with other programs like cubic.

argilo, 2020-10-29:
It would appear that Gqrx is unable to receive samples from your RTL-SDR. Unfortunately, I'm not able to reproduce this problem and I don't know why it would occur, so I won't be able to offer any help. If you learn anything more about the problem, let me know and I can have another look.

kukabu, 2020-10-29:
Hello. Try to reset configuration  - 'gqrx --reset' and run again.

nmaster2042, 2020-10-30:
Tried 'gqrx --reset' ran again but didn't helped.

nmaster2042, 2020-10-31:
OK I found what happened.

It was a permission issue on the ~/.config/gqrx directory that was wrong.

the --reset option wasn't useble until I change permissions to match the current user.

Now all is ok.

Thank you for help.

argilo, 2020-10-31:
Thanks for digging into this @nmaster2042! I was able to reproduce the issue by making `~/.config/gqrx` non-writable. I'll re-open this issue for now, since I think gqrx should at a minimum print an error if it's unable to write to its config file.

good-point-lab, 2025-02-11:
If I would like to start Gqrx with an explicit configuration file, what exact syntasis in the startup dk.gqrx.gqrx.desktop it should be?
Do you know if the quotation needed is correct?  
gqrx -c "~/.config/gqrx/default.conf"

argilo, 2025-02-11:
@good-point-lab For questions of that sort, please use the discussion group: https://groups.google.com/g/gqrx

nmaster2042, 2025-02-11:
Yes it should work, see https://www.mankier.com/1/gqrx

good-point-lab, 2025-02-11:
I will be using the discussion groups, thanks for pointing that out.