gqrx-sdr/gqrx #849: A suggestion for improvement

ulrichloose, 2020-10-28:
I keep stumbling over one thing when using the mouse:

In the pan adapter field the mouse changes into one hand and you can zoom in and out using the scroll wheel. Very convenient! Two things would be nice - the field with the frequency information should be displayed in different color s the Panadaper, it often happens that you want to zoom, but change the frequency with the scroll wheel.

It would be even nicer if you didn't draw the red pointer of the current frequency / bandwidth up to the Waterfall window, but instead ended above the frequency line. Background - I want to zoom in exactly on the frequency, but usually change the bandwidth or the frequency.

I would like to change something myself, but that is beyond my capabilities ...

Maybe worth considering?

ulrichloose, 2020-10-28:
I found the "improvement" in the file plotter.cpp ...

Replace "h" with "aAxisTop" for the length of the Y extension of the filter line and the filter box:

painter.setOpacity(0.3);
painter.fillRect(m_DemodLowCutFreqX, 0, dw, xAxisTop, QColor(PLOTTER_FILTER_BOX_COLOR));
painter.setOpacity(1.0);
painter.setPen(QColor(PLOTTER_FILTER_LINE_COLOR));
painter.drawLine(m_DemodFreqX, 0, m_DemodFreqX, xAxisTop);

and so that the frequency line with the "OpenHandCursur" lies in front of the filter line, the order of the queries in void CPlotter :: mouseMoveEvent (QMouseEvent * event) must be changed:

if (onTag)
            {
                setCursor(QCursor(Qt::PointingHandCursor));
                m_CursorCaptured = BOOKMARK;
            }
            else if (isPointCloseTo(pt.x(), m_YAxisWidth/2, m_YAxisWidth/2))
            {
                if (YAXIS != m_CursorCaptured)
                    setCursor(QCursor(Qt::OpenHandCursor));
                m_CursorCaptured = YAXIS;
                if (m_TooltipsEnabled)
                    QToolTip::hideText();
            }
            else if (isPointCloseTo(pt.y(), m_XAxisYCenter, m_CursorCaptureDelta+5))
            {
                if (XAXIS != m_CursorCaptured)
                    setCursor(QCursor(Qt::OpenHandCursor));
                m_CursorCaptured = XAXIS;
                if (m_TooltipsEnabled)
                    QToolTip::hideText();
            } 
            else if (isPointCloseTo(pt.x(), m_DemodFreqX, m_CursorCaptureDelta))
            {
                // in move demod box center frequency region
                if (CENTER != m_CursorCaptured)
                    setCursor(QCursor(Qt::SizeHorCursor));
                m_CursorCaptured = CENTER;
                if (m_TooltipsEnabled)
                    QToolTip::showText(event->globalPos(),
                                       QString("Demod: %1 kHz")
                                               .arg(m_DemodCenterFreq/1.e3f, 0, 'f', 3),
                                       this);
            }
            else if (isPointCloseTo(pt.x(), m_DemodHiCutFreqX, m_CursorCaptureDelta))
            {
                // in move demod hicut region
                if (RIGHT != m_CursorCaptured)
                    setCursor(QCursor(Qt::SizeFDiagCursor));
                m_CursorCaptured = RIGHT;
                if (m_TooltipsEnabled)
                    QToolTip::showText(event->globalPos(),
                                       QString("High cut: %1 Hz")
                                               .arg(m_DemodHiCutFreq),
                                       this);
            }
            else if (isPointCloseTo(pt.x(), m_DemodLowCutFreqX, m_CursorCaptureDelta))
            {
                // in move demod lowcut region
                if (LEFT != m_CursorCaptured)
                    setCursor(QCursor(Qt::SizeBDiagCursor));
                m_CursorCaptured = LEFT;
                if (m_TooltipsEnabled)
                    QToolTip::showText(event->globalPos(),
                                       QString("Low cut: %1 Hz")
                                               .arg(m_DemodLowCutFreq),
                                       this);
            }
            else

....
[image]

argilo, 2020-10-28:
> I want to zoom in exactly on the frequency, but usually change the bandwidth or the frequency.

That's a good point -- I've trained myself to put my mouse beside the red line while zooming in the frequency axis, but reserving the entire axis for zooming would probably be better.

Since this is a UI change, it would probably be best to put it in the next feature release (i.e. 2.14).

For your other suggestion, I'm curious what exactly you mean:

> the field with the frequency information should be displayed in different color s the Panadaper

Could you annotate a screenshot to show me exactly which UI elements you're referring to?

ulrichloose, 2020-10-29:
I mean the background color behind the frequency display to make the area more clearly in which the OpenHand cursor applies. This only applies if "Fill" is activated in the FFT settings.

Otherwise I've come to the conclusion that it is better to draw the red line to the bottom edge of the pan window. However, the mouse cursor should not change from the OpenHand cursor in the area of the frequency line. This is very easy to do by changing the order of the mouse cursor evaluation. See my last post.

Unfortunately, I was unable to exclude this area from the filling...

[image]

ulrichloose, 2020-10-31:
In the meantime I did a little tinkering and now everything works the way I imagine ...

Just a few small changes need to be made in plotter.cpp:

[patch.txt](https://github.com/csete/gqrx/files/5469090/patch.txt)

[image]

argilo, 2020-11-15:
I opened pull request #867 to change the precedence in `CPlotter::mouseMoveEvent`, which should make it easier to zoom.

The change to the appearance of the frequency axis is a bit more drastic, and I don't think it looks good when the FFT overlaps the frequency axis. I'll leave that out and maybe try to find a better way to highlight the axes in the future.

argilo, 2020-11-28:
@ulrichloose How do you find the interface after the changes I made in #867? Do you still think the appearance of the frequency axis should be changed?

ulrichloose, 2020-11-28:
The solution is absolutely perfect like that! The decisive factor for me is the order and the significantly larger catch area. In any case, I removed my "black box" and use this solution ...

argilo, 2020-12-06:
Thanks for your input. In that case, I'll close off this issue.

closed 2020-12-06 by argilo