gqrx-sdr/gqrx #852: crash when FFT settings waterfall=100%

kgarrels, 2020-11-03:
gqrx will crash when you slide the panadapter/waterfall slider all the way to waterfall.

offending code is in 

```
// Called when screen size changes so must recalculate bitmaps
void CPlotter::resizeEvent(QResizeEvent* )
{
    if (!size().isValid())
        return;

    if (m_Size != size())
    {
        // if changed, resize pixmaps to new screensize
        int     fft_plot_height;

        m_Size = size();
        fft_plot_height = m_Percent2DScreen * m_Size.height() / 100;
        m_OverlayPixmap = QPixmap(m_Size.width(), fft_plot_height);
        m_OverlayPixmap.fill(Qt::black);
        m_2DPixmap = QPixmap(m_Size.width(), fft_plot_height);
        m_2DPixmap.fill(Qt::black);

        int height = m_Size.height() - fft_plot_height;
        if (m_WaterfallPixmap.isNull())
        {
            m_WaterfallPixmap = QPixmap(m_Size.width(), height);
            m_WaterfallPixmap.fill(Qt::black);
        }
        else
        {
            m_WaterfallPixmap = m_WaterfallPixmap.scaled(m_Size.width(), height,
                                                         Qt::IgnoreAspectRatio,
                                                         Qt::SmoothTransformation);
        }

        m_PeakHoldValid = false;

        if (wf_span > 0 && height!=0)   // make sure height is not 0, otherwise crashes
            msec_per_wfline = wf_span / height;
        memset(m_wfbuf, 255, MAX_SCREENSIZE);
    }

    drawOverlay();
    emit newSize();
}
```

argilo, 2020-11-03:
Are you using the latest release of Gqrx? I can't reproduce this. A similar issue was fixed in #848.

argilo, 2020-11-03:
If you're able to reproduce with Gqrx 2.13.3, please let me know the exact steps you followed.

kgarrels, 2020-11-03:
maybe waterfall=100% is confusing...I get the crash like so:
- open fft settings dock
- move the slider all the way to the right, waterfall hight will be 0
- crash

argilo, 2020-11-03:
I still can't reproduce this. Which version of Gqrx are you running?

argilo, 2020-11-04:
I was finally able to reproduce. The issue only occurs when the "Time span" setting is set to something other than "Auto".

argilo, 2020-11-04:
I opened #853 to fix this.

kgarrels, 2020-11-04:
perfect, thanks.

PR #853 (2020-11-05): Fix crash when time span is set and waterfall height is set to 0%
Fixes #852.

To avoid division by zero, `height` should be checked before calculating `msec_per_wfline`. This mirrors the logic that is used in `setWaterfallSpan`:

https://github.com/csete/gqrx/blob/46b5578875051e1ef83b7b46e32230b7220df32c/src/qtgui/plotter.cpp#L525-L526
closed 2020-11-04 by kgarrels