gqrx-sdr/gqrx #854: When Demod Off the filter is still shown after restart

rolnas, 2020-11-05:
When switching to "Demod Off" mode the filter in waterfall is empty initially, but after restart or other changes it shows the last saved filter in waterfall. As a workaround if I remove filter_low_cut and filter_high_cut values in config before starting gqrx it works OK, until I choose again something else as mode and switch back to "Demod Off".

I suggest to introduce filter width as None, to indicate empty filter for the case "Demod Off" of mode or some other solution for this case (possible some more if's in the code).

argilo, 2020-11-06:
I opened #858 to fix this. If you have a chance to test it, I would appreciate it.

closed 2020-11-07 by argilo

rolnas, 2020-11-08:
I did testing with v2.13.5 and it works OK now.
Thanks

PR #858 (2020-11-07): Don't set the filter width if demod is off
Fixes #854.

If "Mode" is set to "Demod Off", then we shouldn't set the filter width during startup.