gqrx-sdr/gqrx #862: Since the last updates something is not worked properly with the settings

ulrichloose, 2020-11-12:
on my Mac...

default.conf

...

[output]
device = pass-thru

is the entry created by version 2.13.3. Pass-Thru is the capture driver from iShowU.

If you start the latest version of gqrx, everything seems to work normally. However, nothing is displayed under Device in the config window - the field remains empty. If you want to change that and just set it to pass-thru, gqrx does not take over that but is tangled up with the entries and writes something wrong in the default.config.

First start:

....
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 4
  0: Built-in input I: 2 O: 0
  1: Built-in Output I: 0 O: 2
  2: iShowU Audio Capture I: 2 O: 2
  3: Pass-Thru I: 0 O: 2
Audio device 'Pass-Thru' has index 3

...

nothing is displayed in the Device menu. Change in the menu to "Pass-Thru" -> restart.

....
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 4
  0: Built-in input I: 2 O: 0
  1: Built-in Output I: 0 O: 2
  2: iShowU Audio Capture I: 2 O: 2
  3: Pass-Thru I: 0 O: 2
Audio device 'iShowU Audio Capture' has index 2

But that's not true, it has to be index 3, because Pass-Thru was selected. After that you can't get any further with the latest version!

argilo, 2020-11-12:
Thanks for the bug report.

@kolbma It looks like this is related to your pull request #774. Would you be able to have a look?

kolbma, 2020-11-12:
@ulrichloose Any chance you could compile with qtdebug enabled and attach the log output? 

But with a working config in file, it should be ok to do no changes in the dialog and cancel the dialog.  

@argilo I've no time for this at the moment. If it is still open in 2-3 weeks I'll dive into it again.

ulrichloose, 2020-11-12:
I've now circled that a little:

From version f9a4a4e6c7201b6271b1a860773c0c26d4b1f12e

update output device list on scan ...
added CIoConfig :: updateOutDev ()

there is an error. From this version the entry "Default" is missing from the displayed list and something wrong is written in the settings file, because only three of four entries are displayed in my case and the index is no longer correct.

I am not an expert, but I don't see any point in this "update" - the main concern of finding a USB device that was inserted later has been fulfilled since the "update input device list". Why then still look for audio output devices that practically do not change? But that also worked up to the last version - why then another update on scan?

kolbma, 2020-11-12:
Well there exist external audio hardware... so the idea has been to enable the detection of these also without restarting gqrx.  

Maybe the problem is only the `ui->outDevCombo->clear();`: [line 637](https://github.com/csete/gqrx/commit/f9a4a4e6c7201b6271b1a860773c0c26d4b1f12e#diff-088fd0ffee00dd5ebc1e4834814fc5f91cc7d4a196a86597ed283d2df7b437afR637)

Could be that there is already a "Default" entry in the UI which isn't added again afterwards.  
Something similar how it is done at [line 688](https://github.com/csete/gqrx/commit/f9a4a4e6c7201b6271b1a860773c0c26d4b1f12e#diff-088fd0ffee00dd5ebc1e4834814fc5f91cc7d4a196a86597ed283d2df7b437afR688)  
So then I'd have missed the re-initialization of the combo.
Need to have a look outside of github.

nmaster2042, 2020-11-12:
I confirm this issue: on my system (ubuntu 20.10), no more Default entry in the audio output device list.
But, It doesn't affect sound because the first audio is the only one I have here.

ulrichloose, 2020-11-12:
I have already thought of the places and added a default device as a test immediately after clear the Combo.
 
ui-> outDevCombo-> clear ();

ui-> outDevCombo-> addItem (m_settings-> value ("output / device", "Default"). toString (),
m_settings-> value ("output / device", "Default"). toString ());

When starting without default.conf, the selection list is correct. The first entry is then "Default" and then the devices follow. The selection is then saved correctly in default.conf and the correct port is selected again at the next start. Small problem - after a restart, "Default" no longer appears as the first entry in the list, but the selected port. So the same entry appears twice.

That's enough for me - I set it once and then never again (;-) Others have to decide whether it can be solved that way.

PR #863 (2020-11-13): Include "Default" entry in audio output device list
Fixes #862.

The change in #774 inadvertently removed the "Default" entry from the audio output device list, which shifted all the other devices up by one and broke device selection. After clearing `outDevCombo`, we need to add back the default entry that is present in ioconfig.ui.

@ulrichloose @kolbma @nmaster2042
closed 2020-11-13 by argilo