gqrx-sdr/gqrx #868: [Feature request] Higher sample rate output for external applications

daniestevez, 2020-11-15:
Currently GQRX has the ability to stream by UDP its audio output. The format is 48ksps real int16 samples. People are using this to connect GQRX to gr-satellites, by using the SSB demodulator in GQRX for BPSK modulations and the FM demodulator for FSK modulations. This works well, but it doesn't allow very high baudrate modulations (such as the 57.6kbaud FSK used by BOBCAT-1) to be sent, since they don't fit in 48ksps.

It would be interesting to have an additional wider bandwidth output that can be used to to send out whatever signal is tuned in GQRX. For flexibility, this output could be complex IQ data. gr-satellites and other external software are capable of performing the required demodulation. GQRX would just be used as a GUI to see the spectrum, click on signals, and maybe Doppler-correct with Gpredict.

As a basis, maybe a 48ksps complex IQ output is a good idea. Some other software similar to GQRX already have the ability to record to WAV file a 48ksps complex IQ centred on the signal the user is tuned to (I keep forgetting if this is done by HDSDR or SDR#, but I think it's one or both of those). This would already allow 2x the current bandwidth to be sent.

Perhaps it would also be nice to be able to select the sample rate of this complex IQ output, maybe limited to power of two multiples of 48ksps (so 48, 96, 192, 384, which are all pretty common sample rates). However I'm not sure how well this fits into the existing DSP of GQRX. I don't know how the SDR samples (which come at pretty much any possible sample rate) get decimated down to 48ksps. If the sample rate is reduced by a sequence of half-band filters, then it would be easy to select the output of any of those. If the sample rate goes down directly by several octaves, then it's not so convenient.

Speaking of the protocol to send the samples, probably UDP is still good. The main use case for this are two applications running on the same computer or over a fast LAN, so there shouldn't be the worry to loose many packets. Perhaps it's good to add a header to the UDP packets with a sequence number and some other metadata, but then this requires external applications to know about this header.

argilo, 2020-11-16:
> As a basis, maybe a 48ksps complex IQ output is a good idea.

This already exists. Select "Raw I/Q" mode in the Receiver Options panel, widen the filter, then click the "..." button in the lower right corner of the Audio panel and enable the Stereo option in the Network tab.

daniestevez, 2020-11-16:
That's quite useful! I vaguely recall that Csete told me he was going to implement that feature, but I never realized he did.

I think that @K4KDR will be interested in checking out this existing feature, since it covers all the immediate use cases he had in mind.

K4KDR, 2020-11-16:
Thank you both for discussion how we can get maximum use from GQRX.  I see that the maximum that I can spread the passband w/ "Mode" set to "Raw I/Q"  is 80k... I don't I don't know if that would be a problem w/ higher rates or not.  Regarding the Audio Panel Network tab, I don't see any reference to "Stereo".  And finally, if the output from GQRX was not demodulated, would passing it to a GNU Radio flowgraph via UDP still work in the same way?  Thanks very much!

[image]

argilo, 2020-11-16:
> the maximum that I can spread the passband w/ "Mode" set to "Raw I/Q" is 80k

The output sample rate is currently limited to 48k, so setting the passband wider than that won't be of any use.

If higher output sample rates are added to gqrx in the future, then the passband will be widened as well.

> I don't see any reference to "Stereo"

This feature was added in gqrx 2.13. From the screenshot, it looks like you're running 2.12.

> would passing it to a GNU Radio flowgraph via UDP still work in the same way?

The I/Q samples are interleaved, so you'll need to deinterleave them and convert to complex with something like Deinterleave + Float to Complex.

csete, 2020-11-18:
Right now, Raw I/Q is just a mode in the narrowband receiver (AM/FM/SSB) and therefore limited to 48 kHz output. In the longer term, it could be implemented as a dedicated receiver, like WFM, and run at higher sample rates. Of course, the audio output to the sound system would have to be disabled when using arbitrary output rates.