gqrx-sdr/gqrx #869: No data with Afedri NETSDR

gvanem, 2020-11-15:
Using the `SoapySdrUtil --make=driver=netsdr` works fine to find my Afedri receiver connect at the LAN (`10.0.0.50:50000`):
```
Make device driver=netsdr
Found hostAddr 10.0.0.50, port 50000.
Using Afedri AFEDRI-SDR SN 31351F39 BOOT 1 FW 106 HW 5
  driver=netSDR
  hardware=NetSDR
```

But using the device string `soapy=netsdr=10.0.0.50:50000` in my `default.conf` throws no error. But no spectrum nor sample-data either. Not sure how is to blame; `gr-osmosdr` or `SoapySDR`. Anyone?

nmaster2042, 2020-11-15:
Did you try with another SDR program directly based on SoapySDR ?

I think the SoapyNetSDR driver is for RFSPACE SDRs, not Alfedri, but I may be wrong.

In source code, no mention of Afedri, only NetSDR, Cloud IQ SDR-IP.

gvanem, 2020-11-15:
> Did you try with another SDR program directly based on SoapySDR ?

No, only tried SDR-Console and SDR-Sharp v1630 with my Afedri (newer versions does not work).

> In source code, no mention of Afedri, only NetSDR, Cloud IQ SDR-IP.

The `SoapyNetSDR` module should support these very similar radios:
```
    RFSPACE_SDR_IQ,  RFSPACE_SDR_IP,
    RFSPACE_NETSDR,  RFSPACE_CLOUDIQ, AFEDRI_SDR_NET
```
(the enum `radio_type`).

nmaster2042, 2020-11-15:
In source code I foud:

enum radio_type
  {
    RADIO_UNKNOWN = 0,
    RFSPACE_SDR_IQ,
    RFSPACE_SDR_IP,
    RFSPACE_NETSDR,
    RFSPACE_CLOUDIQ
  };

Where did you find AFEDRI_SDR_NET ?

gvanem, 2020-11-15:
> Where did you find AFEDRI_SDR_NET ?

Just something I added for later. But the Afedri should be almost equal to a NetSDR radio. So it ought to work in GQrx. 
It works in in SoapySDR I think; `SoapySDRUtil.exe --probe=driver=netsdr`:
```
Probe device driver=netsdr
Found hostAddr 10.0.0.50, port 50000.
getStreamFormats
getNativeStreamFormat

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=netSDR
  hardware=NetSDR

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: NO
  Stream formats: CF32
  Native format: CF32 [full-scale=1]
  Antennas: RX
  Full gain range: [-30, 0] dB
    ATT gain range: [-30, 0] dB
  Full freq range:  MHz
    RF freq range:  MHz
  Sample rates: 0.02, 0.032, 0.04, 0.05, 0.08, ..., 0.625, 0.8, 1, 1.25, 2 MSps
  Filter bandwidths: 34 MHz
```

But a bit strange the `Full freq range: ` is empty and `Supports AGC: NO`.

nmaster2042, 2020-11-15:
OK I understand, you are working at adding Afedri on the SoapyNetSDR driver.

If the protocol is the same as RFSPACE devices it may have some small differences.

It seems you managed the first step: discover the device on the network.

Probably it needs some more mods to work.

The only SDR opensource SDR program supporting afedri is I'm aware of is quisk.

On Aledri download "Quisk support" section you have a module you can get.

It may be a source of info to check.

nmaster2042, 2020-11-15:
There is also the sdr_commander section where you can grab the source code of sdr_commander.

This is the command line utility to control the device, so you could find more details of sent commands, compare with SoapyNetSDR and add what is missing.

gvanem, 2020-11-16:
Building Quisk is impossible for me since:
 * It needs MinGW due to use of C99 complex variables.
 * It needs a 64-bit Python3.

I think I'll concentrate on patching gr-osmosdr instead. Since it already supports RFspace radios.

gvanem, 2020-12-01:
@nmaster2042 
> The only opensource SDR program supporting afedri is I'm aware of is quisk.

I just now tried with [CubicSDR]( https://github.com/cjcliffe/CubicSDR). It works like a charm with my Aferdi. Although the SoapySDR .DLL identifies it as  `RFspace NetSDR SN 31351F39`.  This is the exact same `NetSDRSupport.dll` GQrx load I think. But via `gr-osmosdr`. So a bug there then?
[image]

Shannon Volmet coming in load and clear. But I must says the tuning in CubicSDR is even more confusing and cumbersome than in GQrx.

nmaster2042, 2020-12-04:
Yes, probably issue inside gr-osmosdr if it's ok with native Soapy layer.