gqrx-sdr/gqrx #874: Cannot control gain and turn off AGC

rolnas, 2020-11-17:
Recently I got Airspy HF+ Discovery - it works OK, but in challenge situations I want to turn off AGC and control gain manually.
In 2.13.5 gqrx it is not possible to control both. For RTL-SDRv3 I can control both for I/Q mode (not direct).

P.S. I'm using appimage with libairspyhf library (1.6.8). Firmware was upgraded to latest 3.0.7-cd.

nmaster2042, 2020-11-17:
This is a gr-osmosdr part.

If you want to control gain, you'll need to build a forked version of gr-osmosdr and gqrx.

There is one with improved airspy hf+ support here:

https://github.com/ast/gr-osmosdr/tree/airspyhf

I'm using it with success for some days now.

rolnas, 2020-11-30:
I did succeed partially soapy way via https://github.com/pothosware/SoapyAirspyHF/, but I found, that I cannot control attenuation/AGC on 73.050MHz (transceiver IF), but that works on HF frequencies. Probably hardware limitations :-(