gqrx-sdr/gqrx #876: MacOS Catalina (10.15.5) Release

MCKLtech, 2020-11-19:
Is GQRX compatible with MacOS Catalina? 

I'm having trouble compiling it from source and the latest release and Homebrew are very dated at this point.

argilo, 2020-11-19:
I don't see any reason why it wouldn't be.

Can you post the details of the problem you're encountering?

argilo, 2020-11-28:
CI builds are running fine on macOS 10.15.7, and no details of the error have been provided, so I'll close this off for now.

closed 2020-11-28 by argilo

MCKLtech, 2020-11-28:
Is there a pre-compiled download available of the latest builds? CI builds are great but for those of us that have trouble building from source, it severely limits the accessibility of GQRX. Alternatively, a full end to end tutorial for compiling from source, to include requirements, would be helpful as the current docs take large leaps of assumed prior knowledge.

argilo, 2020-11-28:
Unfortunately there are no pre-compiled builds available for macOS at the moment. I'd like to have GitHub Actions build macOS packages eventually, but this is an area I don't have much expertise in.

MCKLtech, 2020-11-28:
Ok, I included my cmake build log above if anyone can offer any pointers on the build process for MacOS.

The instructions on the installation are also slightly wrong. If you cd to the /build directory and run cmake, you will get:

> CMake Error: The source directory "/Users/XXXX/Downloads/gqrx source/gqrx.git/build" does not appear to contain CMakeLists.txt.
> Specify --help for usage, or press the help button on the CMake GUI.

Above all, not looking for a shortcut or hand out here, I'd be more than happy to contribute a step by step, if someone can help me get it built.

argilo, 2020-11-28:
It looks like you're missing gr-osmosdr. If you're using homebrew, I'd suggest doing `brew install gr-osmodr` before building Gqrx.

argilo, 2020-11-28:
For reference, here's how the CI build works:

https://github.com/csete/gqrx/blob/a86ec4640abec6f1018779c095f4d1ce859a87f3/.github/workflows/ci.yml#L27-L35

I expect these steps should work on your machine:
```
brew install gr-osmodr
git clone https://github.com/csete/gqrx.git
cd gqrx
mkdir build
cd build
cmake ..
make
sudo make install
```

argilo, 2020-11-28:
I opened #889 to track macOS builds as a feature request.

MCKLtech, 2020-11-28:
Thanks, that should get me going in the right direction. Minor bump however:

> brew install gr-osmodr
> ==> Searching for similarly named formulae...
> Error: No similarly named formulae found.
> Error: No available formula or cask with the name "gr-osmodr".
> ==> Searching for a previously deleted formula (in the last month)...
> Warning: homebrew/core is shallow clone. To get complete history run:
>   git -C "$(brew --repo homebrew/core)" fetch --unshallow
> 
> Error: No previously deleted formula found.
> ==> Searching taps on GitHub...
> Error: No formulae found in taps.

MCKLtech, 2020-11-28:
Ok, with  gr-osmosdr installed via brew, this is the cmake log:

> cmake ..
> CMake Warning (dev) at /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:424 (message):
>   The package name passed to `find_package_handle_standard_args` (PkgConfig)
>   does not match the name of the calling package (Gnuradio-osmosdr).  This
>   can lead to problems in calling code that expects `find_package` result
>   variables (e.g., `_FOUND`) to follow a certain pattern.
> Call Stack (most recent call first):
>   /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/FindPkgConfig.cmake:67 (find_package_handle_standard_args)
>   cmake/Modules/FindGnuradio-osmosdr.cmake:1 (INCLUDE)
>   CMakeLists.txt:111 (find_package)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- Checking for module 'gnuradio-osmosdr'
> --   No package 'gnuradio-osmosdr' found
> CMake Warning (dev) at /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:424 (message):
>   The package name passed to `find_package_handle_standard_args`
>   (GNURADIO_OSMOSDR) does not match the name of the calling package
>   (Gnuradio-osmosdr).  This can lead to problems in calling code that expects
>   `find_package` result variables (e.g., `_FOUND`) to follow a certain
>   pattern.
> Call Stack (most recent call first):
>   cmake/Modules/FindGnuradio-osmosdr.cmake:28 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
>   CMakeLists.txt:111 (find_package)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> CMake Warning (dev) at /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:424 (message):
>   The package name passed to `find_package_handle_standard_args` (PkgConfig)
>   does not match the name of the calling package (GMP).  This can lead to
>   problems in calling code that expects `find_package` result variables
>   (e.g., `_FOUND`) to follow a certain pattern.
> Call Stack (most recent call first):
>   /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/FindPkgConfig.cmake:67 (find_package_handle_standard_args)
>   /usr/local/lib/cmake/gnuradio/FindGMP.cmake:1 (include)
>   /usr/local/lib/cmake/gnuradio/FindMPLIB.cmake:1 (find_package)
>   /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
>   /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:26 (find_dependency)
>   CMakeLists.txt:114 (find_package)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> CMake Warning (dev) at /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:424 (message):
>   The package name passed to `find_package_handle_standard_args` (PkgConfig)
>   does not match the name of the calling package (MPIR).  This can lead to
>   problems in calling code that expects `find_package` result variables
>   (e.g., `_FOUND`) to follow a certain pattern.
> Call Stack (most recent call first):
>   /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/FindPkgConfig.cmake:67 (find_package_handle_standard_args)
>   /usr/local/lib/cmake/gnuradio/FindMPIR.cmake:1 (include)
>   /usr/local/lib/cmake/gnuradio/FindMPLIB.cmake:2 (find_package)
>   /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
>   /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:26 (find_dependency)
>   CMakeLists.txt:114 (find_package)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- Checking for module 'mpir >= 3.0'
> --   No package 'mpir' found
> -- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
> CMake Error at /usr/local/Cellar/cmake/3.19.0/share/cmake/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
>   By not providing "FindVolk.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "Volk", but
>   CMake did not find one.
> 
>   Could not find a package configuration file provided by "Volk" with any of
>   the following names:
> 
>     VolkConfig.cmake
>     volk-config.cmake
> 
>   Add the installation prefix of "Volk" to CMAKE_PREFIX_PATH or set
>   "Volk_DIR" to a directory containing one of the above files.  If "Volk"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> Call Stack (most recent call first):
>   /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:46 (find_dependency)
>   CMakeLists.txt:114 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!