gqrx-sdr/gqrx #878: Dragging X frequency axis breaks waterfall untill left-click on it

ankostis, 2020-11-19:
With latest `gqrx-2.13.5-1` build in debian-SID i get a buggy behavior in the waterfal pane: 

When draggingthe frequency to the left, the new part of the waterfall dissappear (becomes black), and a right-click isneeded to begin to be updated: 
[image]

- Has anybody else pobserved that?
- Is this expected?

argilo, 2020-11-19:
I checked and the behaviour is the same in 2.12.1 so it's not a new issue. I agree the behaviour is unintuitive. The frequency bar should probably either prevent you from scrolling past the edge of the FFT, or else adjust the hardware frequency to let you see more.

ankostis, 2020-11-20:
Right-clicking resets X-axis to the previous limits (so frequency had not changed), but clinking sets the offset to the pointed frequency, so there is a code-path that changes the HW-frequency.

argilo, 2020-11-22:
While looking into this, I noticed that dragging the frequency axis with the center mouse button is meant to change the hardware frequency. Unfortunately that feature was broken, but I've fixed it in #883.

Now that that's fixed, I'll experiment with having the left button adjust the hardware frequency when the edge of the FFT is reached.

argilo, 2020-11-28:
There's already an issue for this (#330) so I'll close this.

closed 2020-11-28 by argilo