gqrx-sdr/gqrx #879: Use of .grc files for setting up GNUradio flow

pe1chl, 2020-11-20:
Would it be possible to replace the function calls in GQRX that setup the GNUradio flow by the loading of a (or a couple of) .grc files that one could modify using gnuradio-companion?
It would make it easier to experiment with the SDR core of GQRX, e.g. to add AFC, new demodulators/decoders, etc.

argilo, 2020-11-28:
I don't think it would be practical to have Gqrx load GRC files directly. Parsing a GRC file and dynamically instantiating the necessary blocks would be very complicated.

As of GNU Radio 3.8, GNU Radio Companion can generate C++ code, and I think that would be a more practical way to experiment with Gqrx's DSP.

closed 2020-11-28 by argilo

pe1chl, 2020-11-28:
Ok I (apparently mistakenly) assumed that the parsing and loading of a .grc file would be an available library routine...

My question then changes to the request of providing the existing GNU Radio block structure as a .GRC file so people can experiment with it in GRC and then submit feature implementation ideas to those who understand how to modify the C++ code.