gqrx-sdr/gqrx #886: A Problem with the settings file...

ulrichloose, 2020-11-27:
It has happened to me several times that I accidentally entered an incorrect entry on my RTLSDR stick. Mostly it is a space such as "rtl = 0, direct_samp = 2". It seems to me that a bug has crept in since the innovation to be able to use different settings. In any case, there is a long error message from which you can no longer get out without changing or deleting the settings file manually. In the past you could change the settings in gqrx, today you can only delete or edit externally.

It would be nice if you could bring it back to the old state - deleting the settings is no solution.

gqrx exited with an exception:
bad lexical cast: source type value could not be interpreted as target
More information can be found by launching it from the console.

argilo, 2020-11-28:
I can't reproduce this issue. I can get the "bad lexical cast" message by setting the device string to "rtl=0.0" but I'm always able to edit the configuration after seeing that message.

argilo, 2020-11-28:
I also don't see any difference in behaviour between 2.13.5 and 2.14. Please let me know exactly what steps you're following and where you notice a difference in behaviour between versions.

ulrichloose, 2020-11-28:
I have to apologize here ... I guess I was a little overzealous ...

The problem has apparently been going on longer than I wrongly suspected.

For a precise description of the "inconsistencies":

gqrx is running, rtlsdr is set to `rtl=0,direct_samp=0` and I want to switch to `rtl=0,direct_samp= 2` Unfortunately, I make a mistake and accidentally put a space in front of the 2 in = 2. Result "exited with exception", in default.conf it remains with a "crashed = false" and gqrx is terminated.

At the next start the same again "exited with exception" but fortunately a crashed = false is written in the default.cfg and thus you get the possibility to correct the wrong entry in gqrx with the second start attempt ...

Obviously I was a bit overzealous and made the "second start attempt" with an older version of gqrc and wrongly assumed that the problem would not arise with the old version.

The problem has existed for a longer time, but it is much smaller, since the entries can be corrected on the second start attempt.

argilo, 2020-11-28:
Thanks for the clarification. I was able to reproduce the issue with `rtl=0,direct_samp= 2`.

argilo, 2020-11-28:
This should be fixed in a86ec4640abec6f1018779c095f4d1ce859a87f3.

closed 2020-11-28 by argilo