gqrx-sdr/gqrx #889: Automated macOS builds

argilo, 2020-11-28:
It would be nice to build macOS releases (DMG files?) automatically using GitHub Actions. Unfortunately I have little experience building macOS packages, so help would be appreciated.

ulrichloose, 2020-11-28:
Unfortunately, this is a bit more complicated with the Mac if you do not assume that the user has all the dependencies on the computer.

You then need an extra tool that copies all dylibs into the app bundle and patches the actual program gqrx to the new positions of the dylibs. In addition, of course, a run of macdeployqt so that the dependencies and paths are copied and patched and all the little helper programs for the individual SDR hardware drivers must also be supplied.

There was a DMG file here version 2.11? which made the problem clear. Whether this can be automated - I have my doubts.

In principle, locally - at least that's how I do it - as follows:

dylibbundler -od -b -x ./Gqrx.app/Contents/MacOS/gqrx -d ./Gqrx.app/Contents/Libs/

copies all dependencies into the app framework and changes the paths in gqrx

macdeployqt ./Gqrx.app -no-strip -always-overwrite

does the same with the QT dependencies.

Then you would have to / should sign the result for which you need a corresponding certificate.

Personally, I would think it makes more sense to publish an updated gqrx.app that can be used immediately.

If another Mac user is interested, he can try out the following file (it is signed). All dependencies are included in the program, but no helper programs for the individual SDR hardware (yet) ...

http://ul.to/n6w6i01l

csete, 2020-11-29:
It looks like dylibbundler is the tool I was missing when I was building app bundles. Though I am wondering whether it resolves dependencies recursively, i.e. gqrx depends on gnuradio and gr-osmosdr, gr-osmosdr depends on gnuradio and the driver libraries, which in turn depend on libusb.

The command line tools I used to include in the app bundle usually only depend on the driver library and can be easily patched with install_name_tool

ulrichloose, 2020-11-29:
As far as I can tell, dylibbundler works completely recursively and integrates all required dylibs. Anyway, I checked that with [Macdependency](https://github.com/kwin/macdependency) and also with the mac "Activity Monitor" - which is a bit difficult when you have all the dependencies local.
[](https://github.com/kwin/macdependency)
That's why the attached zip for testing on a Mac that has as few local dependencies as possible.

effeerre75, 2020-11-30:
I was looking at this script but not tested yet
https://gist.github.com/sigsergv/82925dc88e351078d42b
If someone need help fort testing the package no problem 😉

ulrichloose, 2020-12-01:
I killed that with a primitive shell script.

Macdeployqt from the QT5 package is already available on the system. In addition, only the mac dylibbundler has to be installed - you can find it here on github.
 
The only cliff that has to be avoided - in my case, homebrew icu4c does not link to / usr / local / lib and is keg-only because Apple supplies part of icu4c itself. The path to it must be communicated to dylibbundler via -s, otherwise there will be constant inquiries about the storage location.

In any case, it works quite well and creates an app bundle with all dependencies within 20 seconds.

The last line can also be replaced by

macdeployqt Gqrx.app -dmg -no-strip -always-overwrite -verbose = 3

Then a dmg file will be generated.

----------------

[macbundler.zip](https://github.com/csete/gqrx/files/5621136/macbundler.zip)

argilo, 2020-12-03:
Thanks for your help everyone! I was able to get a macOS build going in 6798642ee17cd7523dd29617ba190ce33c41fa05. The output from the first run can be downloaded here: https://github.com/csete/gqrx/actions/runs/397594182

Note: it will only run on macOS 10.15 and later. I tried to add support for older versions by setting `CMAKE_OSX_DEPLOYMENT_TARGET`, but the library dependencies (installed via Homebrew) are built for 10.15, and GitHub Actions does not support older versions of macOS.

argilo, 2020-12-03:
I'll close this off since automated builds are now working. But feel free to leave any feedback here.

closed 2020-12-03 by argilo