gqrx-sdr/gqrx #891: Apply FFT averaging settings to audio pandapter

shuyuan-liu, 2020-11-30:
Currently the averaging slider in the FFT tab only affects the IQ pandapter and has no effect on the audio pandapter. It would be good if the setting also affects the audio pandapter; or add a separate averaging setting for audio?

shuyuan-liu, 2020-11-30:
Did some experiment, copied averaging code from `MainWindow::iqFftTimeout()` to `MainWindow::audioFftTimeout()`. Also created separate buffers so instead of `d_iirFftData[]` there are now `d_iirIqFftData[]` and `d_iirAudioFftData[]`.

Turned out working but not so good: the refresh rate of the audio waterfall seems a lot slower than that of the IQ waterfall, and the averaging makes the audio waterfall look almost all flat. Decreasing the timer interval for audio FFT can solve the issue, but I think it's better just to havea separate averaging slider in the audio settings than doing all this mess.

csete, 2020-11-30:
Remember that audio sample rate is only 48 kHz so you have to trade-off between resolution and refresh rate, and for the same reason, the running average will not have the same visual effect as on the IQ FFT where the sample rate is in most cases much higher than what is needed for FFT.

closed 2020-12-01 by shuyuan-liu