gqrx-sdr/gqrx #899: gqrx crashes

ulrichloose, 2020-12-05:
with the following Message:

gqrx exited with an exception:
port number 1 exceeds max of 0
More information can be found by launching it from the console.

if I use RAW I/Q as demodulator.

From console there is no further information shown...

I use a mac with an RTL-SDR Stick and iShowU audio capture driver. I´m not shure, but I think this behavior is new...

argilo, 2020-12-05:
I'm unable to reproduce the crash. Can you let me know exactly which steps you followed? Which input device are you using? Which buttons did you press after starting Gqrx?

argilo, 2020-12-05:
> A small suggestion for improvement ...

Please open a separate issue for this. Otherwise it is very difficult to track what work remains to be done.

ulrichloose, 2020-12-05:
I have to apologize for the second time ...

Apparently the last update bites with one of my own extensions that I save from update to update using Git Stash ... Mea Culpa.

closed 2020-12-05 by ulrichloose