gqrx-sdr/gqrx #900: MacOS build is missing hardware support

argilo, 2020-12-05:
The automated MacOS build uses gr-osmosdr from Homebrew:

https://github.com/csete/gqrx/blob/2cdb903cec6883e9892c83239ada4bac6bb8fad0/.github/workflows/build.yml#L84-L85

Unfortunately, this package is missing support for the following hardware types:

* Funcube Dongle
* BladeRF
* Airspy HF
* SoapySDR
* FreeSRP

The best option would probably be to improve the Homebrew package to add support for those hardware devices. Alternately, we could build hardware drivers and gr-osmosdr from source in the build.

argilo, 2020-12-05:
To track progress on this:

- [ ] Funcube Dongle
- [x] BladeRF - added in e21ddbaeac71c6d4aee73c24317d3cbd5e93f400
- [x] Airspy HF - added in e21ddbaeac71c6d4aee73c24317d3cbd5e93f400
- [ ] SoapySDR
- [ ] FreeSRP

ulrichloose, 2020-12-08:
I have read to myself that you should avoid installing with sudo on the Mac if possible. If Homebrew is already available, the target folder / usr / local can also be reached without sudo and you can install the two locally compiled drivers without sudo using make install. So to be on the safe side, the sudo could be left out ...

argilo, 2020-12-08:
You're right. I removed the unnecessary `sudo` in faacc796752d5087ea018ab4f362e46423bfe107.

ulrichloose, 2020-12-18:
SoapySDR can be installed via homebrew if you `brew tap pothosware/homebrew-pothos`

`brew search soapy`
==> Formulae
pothosware/pothos/pothossoapy           pothosware/pothos/soapybladerf          pothosware/pothos/soapyosmo             pothosware/pothos/soapyrtlsdr
pothosware/pothos/soapyairspy           pothosware/pothos/soapyhackrf           pothosware/pothos/soapyplutosdr         pothosware/pothos/soapysdr
pothosware/pothos/soapyairspyhf         pothosware/pothos/soapyiris             pothosware/pothos/soapyredpitaya        pothosware/pothos/soapyuhd
pothosware/pothos/soapyaudio            pothosware/pothos/soapynetsdr           pothosware/pothos/soapyremote