gqrx-sdr/gqrx #904: No audio in version 2.14 - MacOS

RobertoD91, 2020-12-11:
On macos 11.0.1 there is no audio with last version of gqrx

versione 2.11.5 work well
2.14.2 and 2.14.3 don't work

tested with rtl-sdr and hackrf

darkworon, 2020-12-11:
I have same issue.

```
➜  ~ gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 5
  0:  Внешние наушники  I:0  O:2
  1:  Микрофон MacBook Pro  I:1  O:0
  2:  Динамики MacBook Pro  I:0  O:2
  3:  Loopback input  I:2  O:2
  4:  Loopback output  I:2  O:2
Using default audio device
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
||PaMacCore (AUHAL)|| OpenStream @ 48000 returned: -9986: Internal PortAudio error
portaudio_sink(): Audio output device does not support requested format.
BandPlanFile is /Users/oleg/.config/gqrx/bandplan.csv
BookmarksFile is /Users/oleg/.config/gqrx/bookmarks.csv
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 5
  0:  Внешние наушники  I:0  O:2
  1:  Микрофон MacBook Pro  I:1  O:0
  2:  Динамики MacBook Pro  I:0  O:2
  3:  Loopback input  I:2  O:2
  4:  Loopback output  I:2  O:2
[INFO] [UHD] Mac OS; Clang version 12.0.0 (clang-1200.0.32.2); Boost_107400; UHD_3.15.0.0-0-unknown
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf redpitaya 
Using AirSpy NOS v1.0.0-rc10-6-g4008185 2020-05-08, samplerates: 2.5M 10M 
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 5
  0:  Внешние наушники  I:0  O:2
  1:  Микрофон MacBook Pro  I:1  O:0
  2:  Динамики MacBook Pro  I:0  O:2
  3:  Loopback input  I:2  O:2
  4:  Loopback output  I:2  O:2
Using default audio device
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
||PaMacCore (AUHAL)|| OpenStream @ 48000 returned: -9986: Internal PortAudio error
portaudio_sink(): Audio output device does not support requested format.
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
portaudio_sink::start(): Failed to open audio stream: Internal PortAudio error
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
```

RobertoD91, 2020-12-11:
Audio recording in .wav is ok and i can play it in vlc.

RobertoD91, 2020-12-11:
Build from souce, same issue

on terminal:

```
xxx@xxx:~/Desktop/git/gqrx.git/build/src$ ./gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
gr::fft: can't import wisdom from /Users/xxx/.gr_fftw_wisdom
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 3
  0:  Microfono MacBook Pro  I:1  O:0
  1:  Altoparlanti MacBook Pro  I:0  O:2
  2:  Loopback Audio  I:2  O:2
Using default audio device
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
||PaMacCore (AUHAL)|| OpenStream @ 48000 returned: -9986: Internal PortAudio error
portaudio_sink(): Audio output device does not support requested format.
BandPlanFile is /Users/xxx/.config/gqrx/bandplan.csv
BookmarksFile is /Users/xxx/.config/gqrx/bookmarks.csv
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 3
  0:  Microfono MacBook Pro  I:1  O:0
  1:  Altoparlanti MacBook Pro  I:0  O:2
  2:  Loopback Audio  I:2  O:2
[INFO] [UHD] Mac OS; Clang version 12.0.0 (clang-1200.0.32.27); Boost_107400; UHD_4.0.0.0-0-unknown
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 3
  0:  Microfono MacBook Pro  I:1  O:0
  1:  Altoparlanti MacBook Pro  I:0  O:2
  2:  Loopback Audio  I:2  O:2
Using default audio device
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
||PaMacCore (AUHAL)|| OpenStream @ 48000 returned: -9986: Internal PortAudio error
portaudio_sink(): Audio output device does not support requested format.
[R82XX] PLL not locked!
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
portaudio_sink::start(): Failed to open audio stream: Internal PortAudio error
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
(...)
```

cmestewa, 2020-12-12:
I also can reproduce on Big Sur. For the originator do you see on Big Sur or an earlier release?

`cstewart@iMac ~ % uname -a
Darwin iMac.local 20.2.0 Darwin Kernel Version 20.2.0: Wed Dec  2 20:39:59 PST 2020; root:xnu-7195.60.75~1/RELEASE_X86_64 x86_64
cstewart@iMac ~ % gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf redpitaya 
Resampling audio 96000 -> 48000
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 2
  0:  Built-in Microphone  I:2  O:0
  1:  Built-in Output  I:0  O:2
Using default audio device
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
||PaMacCore (AUHAL)|| OpenStream @ 48000 returned: -9986: Internal PortAudio error
portaudio_sink(): Audio output device does not support requested format.
BandPlanFile is /Users/cstewart/.config/gqrx/bandplan.csv
BookmarksFile is /Users/cstewart/.config/gqrx/bookmarks.csv
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 2
  0:  Built-in Microphone  I:2  O:0
  1:  Built-in Output  I:0  O:2
[INFO] [UHD] Mac OS; Clang version 12.0.0 (clang-1200.0.32.2); Boost_107400; UHD_3.15.0.0-0-unknown
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Enabled direct sampling mode, input 2
PortAudio V19.6.0-devel, revision 396fe4b6699ae929d3a685b3ef8a7e97396139a4 (version 1246720)
Number of audio devices: 2
  0:  Built-in Microphone  I:2  O:0
  1:  Built-in Output  I:0  O:2
Audio device 'Built-in Output' has index 1
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
||PaMacCore (AUHAL)|| OpenStream @ 48000 returned: -9986: Internal PortAudio error
portaudio_sink(): Audio output device does not support requested format.
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
portaudio_sink::start(): Failed to open audio stream: Internal PortAudio error
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer
portaudio_sink::work(): Error writing to audio device: Invalid stream pointer`

richardh538, 2020-12-13:
I have the same problem!

argilo, 2020-12-14:
@richardh538 Are you also running Big Sur?

I do not have a machine running Big Sur and I can't reproduce the issue, so I'm going to need help diagnosing this.

richardh538, 2020-12-14:
@argilo Yes, I run Big Sur. I can help you with diagnosing. What do you want me to do?

RobertoD91, 2020-12-14:
@cmestewa Yes, on Big Sur

Tested also in a macos 10.15 vm, and sound is ok

@argilo Can i do something? If you want, i can give you access to a VM with Big Sur

NoelM, 2020-12-15:
I am experiencing the same problem, no sound! GQRX 2.14.3 running on macOS Big Sur 11.0.1

In addition, the waterflow disappeared... Have you noticed it?

argilo, 2020-12-15:
Does it work on Big Sur if you build using the Gr-audio backend instead of Portaudio? You can try that by replacing the `cmake ..` line in the build instructions with this:

```
cmake -DOSX_AUDIO_BACKEND:STRING=Gr-audio ..
```

RobertoD91, 2020-12-15:
> Does it work on Big Sur if you build using the Gr-audio backend instead of Portaudio? You can try that by replacing the `cmake ..` line in the build instructions with this:
> 
> ```
> cmake -DOSX_AUDIO_BACKEND:STRING=Gr-audio ..
> ```

It works!

argilo, 2020-12-15:
@RobertoD91 Thanks for checking. That at least provides a workaround for now, and suggests that the problem is related to Portaudio.

argilo, 2020-12-16:
Another question: When running the Gqrx 2.14.3 release (or building with the default Portaudio audio backend), does it help if you select something other than "Default" as the audio output device in the "Configure I/Q devices" dialog?

argilo, 2020-12-16:
It seems like Portaudio may have problems with Big Sur: https://github.com/PortAudio/portaudio/issues/218#issuecomment-731302795

I tried building Gqrx using the master branch of Portaudio, so I'd appreciate if Big Sur users could test this to see if it works any better: https://github.com/argilo/gqrx/actions/runs/424618401

cmestewa, 2020-12-16:
@argilo This new build with the master branch of Portaudio is working for me on macOS 11.1 (20C69)

chpalm, 2020-12-16:
> It seems like Portaudio may have problems with Big Sur: PortAudio/portaudio#218 (comment) <https://github.com/PortAudio/portaudio/issues/218#issuecomment-731302795>

Yes Gqrx-macos-424618401 image works, at least on my Mac mini (2018) with BigSur v.11.1.

/Christer

”in the beginning there was nothing and even that exploded”

NoelM, 2020-12-16:
> It seems like Portaudio may have problems with Big Sur: [PortAudio/portaudio#218 (comment)](https://github.com/PortAudio/portaudio/issues/218#issuecomment-731302795)
>
> I tried building Gqrx using the [...]

Thank you! It works perfectly on macOS 11.1

argilo, 2020-12-16:
Thanks for testing everyone. I'll put out an official release with this change shortly.

I hope Portaudio will make an official release so I can stop using their master branch.

RobertoD91, 2020-12-16:
> Another question: When running the Gqrx 2.14.3 release (or building with the default Portaudio audio backend), does it help if you select something other than "Default" as the audio output device in the "Configure I/Q devices" dialog?

No effect, same issue

K0HYD, 2020-12-24:
> @argilo This new build with the master branch of Portaudio is working for me on macOS 11.1 (20C69)

Just tested this on MacOS 11.1 running on a 2017 iMac with 48 GB of memory. It works great! The last release I found 2.14.3 crashes when I launch it. Is the URL to find your new release version when your release it? Thanks for finding this solution. 73 de Dale - K0HYD

richardh538, 2021-01-01:
Tested with 

> Another question: When running the Gqrx 2.14.3 release (or building with the default Portaudio audio backend), does it help if you select something other than "Default" as the audio output device in the "Configure I/Q devices" dialog?

Tried with my Bose Soundsport (BT). 

Number of audio devices: 4
  0:  dumdumdan  I:1  O:0
  1:  dumdumdan  I:0  O:2
  2:  Built-in Microphone  I:2  O:0
  3:  Built-in Output  I:0  O:2
Audio device 'dumdumdan' has index 1
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
||PaMacCore (AUHAL)|| OpenStream @ 48000 returned: -9986: Internal PortAudio error
portaudio_sink(): Audio output device does not support requested format.
||PaMacCore (AUHAL)|| Error on line 1316: err='-66748', msg=Unknown Error
portaudio_sink::start(): Failed to open audio stream: Internal PortAudio error
/usr/local/bin/gqrx: line 2: 19870 Segmentation fault: 11  '/Applications/Gqrx.app/Contents/MacOS/gqrx' "$@"

lcd15000, 2021-03-30:
Same issue here gqrx 2.14.4 on Big Sur 11.2.2 no sound on "Bose NC 700 HP"

built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf redpitaya 
Using HackRF One with firmware 2018.01.1
PortAudio V19.6.0-devel, revision unknown (version 1246720)
Number of audio devices: 8
  0:  Bose NC 700 HP  I:1  O:0
  1:  Bose NC 700 HP  I:0  O:1
  2:  DELL U2718Q  I:0  O:2
  3:  HP Z27n G2  I:0  O:2
  4:  HP Z27n G2  I:0  O:2
  5:  Unknown USB Audio Device  I:2  O:0
  6:  MacBook Pro Microphone  I:1  O:0
  7:  MacBook Pro Speakers  I:0  O:2
Audio device 'Bose NC 700 HP' has index 1
portaudio_sink(): Audio output device does not support requested format.

argilo, 2022-01-29:
Closing, since this issue was fixed in version 2.14.4. If any audio issues remain in the latest release, a new issue should be opened since they are unrelated to the original issue reported here.

closed 2022-01-29 by argilo