gqrx-sdr/gqrx #908: Changing filter shape resets custom filter width to "normal"

shuyuan-liu, 2020-12-19:
If I drag the filter edges on the pandapter to adjust the filter width, then change the filter shape in the Receiver Options tab, the filter width gets reset to what you would get when selecting width "normal". Despite this, the filter width dropdown box still shows the previous "User (...k)" filter width.

## Screenshots

Starting GQRX with Normal filter width and Normal filter shape:

[image]

Drag on filter edges to increase / decrease filter width. The filter width value in the Receiver Options tab is updated properly.

[image]

Change filter shape to sharp / soft. The actual filter width gets reset, both on the pandapter and as I can tell from the audio output. However the filter width value in the Receiver Options tab is not updated.

[image]

shuyuan-liu, 2020-12-19:
This only happens when the filter width is a "user" value. If the filter width is one of "wide", "normal" or "narrow" selected from the Filter width dropdown, it does not get reset when changing filter shape.

shuyuan-liu, 2020-12-20:
I had a look at the code and my current impression is that a fix might involve more than adding a few lines of code.

When a new filter shape is selected, the filter shape box calls `DockRxOpt::on_filterCombo_activated()`, emitting `DockRxOpt::demodSelected()` to call `MainWindow::selectDemod()`.

`MainWindow::selectDemod()` uses `DockRxOpt::getFilterPreset()` to get the current filter width. The latter thinks a "custom" filter width (index 3) is invalid:

```
void DockRxOpt::getFilterPreset(int mode, int preset, int * lo, int * hi) const
{
    ...
    else if (preset < 0 || preset > 2)
    {
        qDebug() << __func__ << ": Invalid preset:" << preset;
        preset = FILTER_PRESET_NORMAL;
    }
    ...
}
```

This is not the main problem and can be fixed by simply changing the `> 2` to `> 3`. The main problem is I haven't been able to find a way for this function to know the current filter width and then return it. The filter width box knows nothing about the custom filter width; the receiver object should know about it but doesn't have a call to get the filter width either.

I might need to write a call for that in the receiver class, but I'm quite lost currently in reading all the code so this might take a while. Any insights on what I should be looking at?

ulrichloose, 2020-12-25:
I'm not a programmer, just an interested layman ...

This is called "quick & dirty" what I tried ? (;-)

dockrxopt.cpp

https://github.com/csete/gqrx/blob/54aa8b866f42144e3614a7362028a82e1f03c462/src/qtgui/dockrxopt.cpp#L29

int glo, ghi = 0;

..
..
https://github.com/csete/gqrx/blob/54aa8b866f42144e3614a7362028a82e1f03c462/src/qtgui/dockrxopt.cpp#L257-L259

save lo, hi to glo, ghi
glo = lo;
ghi = hi;

https://github.com/csete/gqrx/blob/54aa8b866f42144e3614a7362028a82e1f03c462/src/qtgui/dockrxopt.cpp#L369-L373

else if (preset <0 || preset> 3)

and after line 373

if (preset == 3)
{
* lo = glo;
* hi = ghi;
}
else
{
* lo = filter_preset_table [mode] [preset] [0];
* hi = filter_preset_table [mode] [preset] [1];
}

Then it seems to work ... But only as a dry run because I'm sitting at my laptop and don't have an SDR stick on the device.

ulrichloose, 2020-12-26:
I think that's a bit too cumbersome for a simple user-selectable filter bandwidth. The settings would have to be greatly expanded for the entire table with its own values ​​alone, although the last selected filter bandwidth is already saved and is also assigned to the user filter entry when it is read in at the start, if it deviates from the standard.

From my point of view, it is more disturbing that user can be selected if no custom bandwidth has yet been set. I had thought about graying out the entry User until a value has been stored so that the user makes sense at all.

For that you would have to add to dockrxopt.cpp

```
/ ** Read receiver configuration from settings data. * /
void DockRxOpt :: readSettings (QSettings * settings)
{
```

the entry

```
/ ** Disable Customfield in filterCombo. * /
qobject_cast <QStandardItemModel *> (ui-> filterCombo-> model ()) -> item (3) -> setEnabled (false);
```

inserted to switch off the Custom field for the moment.

and at

`void DockRxOpt :: setFilterParam (int lo, int hi)
`
you have to add

```
    if (filter_index == FILTER_PRESET_USER)
    {
        float width_f;
        width_f = fabs ((hi-lo) /1000.f);
        ui-> filterCombo-> setItemText (FILTER_PRESET_USER, QString ("User (% 1 k)"). arg (width_f));
        qobject_cast <QStandardItemModel *> (ui-> filterCombo-> model ()) -> item (3) -> setEnabled (true);
    }

```
so that Custom field is available for selection. This also happens when reading the settings if a custom bandwidth was last selected.

I find this impressively simple and it does more justice to the actual purpose of a single custom filter bandwidth and does not need a change in the settings ....

But this is just a suggestion to simplify things a little.

shuyuan-liu, 2020-12-26:
@ulrichloose Thanks for the idea, yes I should have thought of that!

ulrichloose, 2020-12-26:
I think that's a bit too cumbersome for a simple user-selectable filter bandwidth. The settings would have to be greatly expanded for the entire table with its own values ​​alone, although the last selected filter bandwidth is already saved and is also assigned to the user filter entry when it is read in at the start, if it deviates from the standard.

From my point of view, it is more disturbing that user can be selected if no custom bandwidth has yet been set. I had thought about graying out the entry User until a value has been stored so that the user makes sense at all.

For that you would have to add to dockrxopt.cpp

```
/ ** Read receiver configuration from settings data. * /
void DockRxOpt :: readSettings (QSettings * settings)
{
```

the entry

```
/ ** Disable Customfield in filterCombo. * /
qobject_cast <QStandardItemModel *> (ui-> filterCombo-> model ()) -> item (3) -> setEnabled (false);
```

inserted to switch off the Custom field for the moment.

and at

`void DockRxOpt :: setFilterParam (int lo, int hi)
`

you have to add

```
    if (filter_index == FILTER_PRESET_USER)
    {
        float width_f;
        width_f = fabs ((hi-lo) /1000.f);
        ui-> filterCombo-> setItemText (FILTER_PRESET_USER, QString ("User (% 1 k)"). arg (width_f));
        qobject_cast <QStandardItemModel *> (ui-> filterCombo-> model ()) -> item (3) -> setEnabled (true);
    }

```
so that Custom field is available for selection. This also happens when reading the settings if a custom bandwidth was last selected.

I find this impressively simple and it does more justice to the actual purpose of a single custom filter bandwidth and does not need a change in the settings ....

But this is just a suggestion to simplify things a little.

Furthermore there is a small error if using bookmarks. They are overriding the global variable for storing the custom filter bandwidth.

The position of the 

```
user_filter_lo = lo;
user_filter_hi = hi;
```

should go to

```
if (filter_index == FILTER_PRESET_USER)
    {
        float width_f;
        user_filter_lo = lo;
        user_filter_hi = hi;
```

so that they are only updated if the filter_index is FILTER_PRESET_USER

ulrichloose, 2020-12-27:
The problem with a recall from the bookmarks has nothing to do with these improvements. If a filter is set asymmetrically, it is temporarily stored and correctly stored as Lo and Hi in the settings. In the bookmarks only the total bandwidth is saved. If a filter is asymmetrically set e.g. -850 500 and activated via "User (1.35K)", a recall from the bookmarks leads to an error and the bandwidth stored in the bookmarks is not set. It stays with the user - 850 500.

The cause can be found in mainwindow.cpp:

https://github.com/csete/gqrx/blob/54aa8b866f42144e3614a7362028a82e1f03c462/src/applications/gqrx/mainwindow.cpp#L2146-L2172

this should be changed to

```
    if(lo >= 0 && hi >= 0) // USB
    {
        hi = lo + bandwidth;
    }
    else if(lo <= 0 && hi <= 0) // LSB
    {
        lo = hi - bandwidth;
    }
    else  // symmetric and anything else
    {
        lo = -bandwidth / 2;
        hi =  bandwidth / 2;
    }
```

shuyuan-liu, 2020-12-27:
Yes @ulrichloose I guess you're right, I did complicate things a bit. How about you make a pull request with your simpler solution that would close this issue? I'll go work on the extra stuff separately.

ulrichloose, 2020-12-27:
Ummpf .... This is the first time in my live to make a pull request (;-) I hope I will be successful..