gqrx-sdr/gqrx #914: DC remove button no longer works.

drmpeg, 2020-12-20:
Toggling the DC remove button has no effect. I've bisected this down to commit https://github.com/csete/gqrx/commit/93bcfae3bf3eb59e49dacfdc036647e447e20fbe

I'll guess the other buttons need to be tested also.

argilo, 2020-12-20:
The DC remove checkbox is working fine for me and 93bcfae3bf3eb59e49dacfdc036647e447e20fbe appears unrelated.

Can you provide more details about your setup? What operating system are you running on? What input device are you using? Which versions of the dependencies are you using?

drmpeg, 2020-12-20:
Pretty ancient stuff. Ubuntu 16.04, GNU Radio 3.7.14. I'm using a bladeRF that has a huge DC spike.

I also don't have gr-iqbal if that makes a difference.

drmpeg, 2020-12-20:
If I comment out the lines:

    if (demod == d_demod)
        return ret;

in receiver.cpp, it works.

drmpeg, 2020-12-20:
I put a printf() in ```receiver::set_demod()```, and it gets called every time I toggle the checkbox. This is while the DSP is running.

argilo, 2020-12-20:
Aha, here's the problem:

https://github.com/csete/gqrx/blob/e1fa944518bd1dc945bc4e2e870aeceb7cb1a32c/src/applications/gqrx/receiver.cpp#L478-L482

argilo, 2020-12-20:
The `d_demod = RX_DEMOD_OFF;` is meant to force `set_demod` to reconfigure the receiver, but that won't happen if the receive mode is already set to "Demod Off".

drmpeg, 2020-12-20:
Yes, I'm in "Demod Off" mode.

argilo, 2020-12-20:
OK. That code is an ugly hack. I'll think about how to do this the right way.

drmpeg, 2020-12-20:
Cool.

argilo, 2020-12-21:
Fixed in 3cc96629a1b7a2a8d88c459ba006710f297395da.

At first I tried adding a `set_enabled` method to `dc_corr_cc` class to allow it to be switched on and off, but hierarchical blocks don't allow the source pad to be connected directly to the sink pad. :frowning_face:

closed 2020-12-21 by argilo