gqrx-sdr/gqrx #915: HD Radio (NRSC5) Support

leoherzog, 2020-12-21:
It would be lovely if we could decode North American HD Radio in GQRX. Reference implementation available at [theori-io/NRSC5] (https://github.com/theori-io/NRSC5).

harbnger, 2023-06-26:
I second this idea, should just take a code pull/merge.
Albertb has a fork here that supports NRSC5:
https://aur.archlinux.org/packages/gqrx-nrsc5-git

argilo, 2023-06-26:
I'd love to include HD Radio support, but there are still active patents covering HD Radio as well as the HDC codec. Perhaps we can look at this again once they've expired (around 2026, I believe).

ve2tmq, 2024-12-19:
Hi folks.  This is what I did.  I take source forked from (https://github.com/Albert-S-Briscoe/gqrx/tree/nrsc5) and compare diff with orig code (it's based on 2.15.9).  Once I've got diff I create patch file.  After I clone code from recent version (2.17.6) and I applied patch.
[nrsc5.patch.txt](https://github.com/user-attachments/files/18190758/nrsc5.patch.txt)

```
cd gqrx
patch -p1 ../nrsc5.patch
mkdir build
cd build
cmake ..
make
```
[image]

geerlingguy, 2025-01-22:
As we're in 2025 and nearing 2026 (I believe the original patents were filed around 2005?), I was looking at the patent issues... and noticed there are like 40 patents covering aspects of 'NRSC-5' (fun umbrella term for all their HD Radio proprietary junk): https://www.nrscstandards.org/standards-and-guidelines/documents/standards/nrsc-5-e/Xperi2023.pdf

Is there a particular patent for reception decoding that is the one lynchpin preventing `nrsc5` from being acceptable in the wider SDR ecosystem? Or is it possible Xperi is going to try to prevent any use of HD Radio signals [through their giant and expanding patent moat](https://www.engineeringradio.us/blog/2010/11/wait-until-the-patent-expires/)?

argilo, 2025-01-22:
At this point, I suspect that enough of the NRSC-5 patents have expired that it should be possible to make an unencumbered demodulator. (Checking over the long list of still-valid patents might be a chore though!)

But there's a second issue at play, namely patents on the HE-AAC audio codec. I don't have a list of HE-AAC patents handy, and I'm not sure exactly when the last relevant patent would expire. Anyone have ideas on that?

leoherzog, 2025-01-23:
@argilo is there any precedent in GQRX for having demodulation logic live within the application, but then outputting/exposing a stream that another third-party decoder like ffmpeg or VLC can take in and process externally?

argilo, 2025-01-23:
@leoherzog There's nothing quite like that in Gqrx today. The closest it can do is expose demodulated audio over UDP.

Unfortunately I don't think existing tools like ffmpeg or VLC can decode the NRSC-5 variant of HE-AAC (called HDC). nrsc5 carries around a patchset to add HDC decode support to faad2 (https://github.com/theori-io/nrsc5/blob/master/support/faad2-hdc-support.patch), and I have patched HDC encode support into fdk-aac (https://github.com/argilo/fdk-aac/tree/hdc-encoder) for use with gr-nrsc5. Upstreaming these changes might be a good starting point. Debian in particular ships a faad2 package, so it might be possible to get HDC decode support into it. Unfortunately I know little about audio codecs so I'm not sure how correct these patches are. (I know for sure that they don't work correctly for all AM streams, and they don't support separate "core" and "enhanced" streams.)