gqrx-sdr/gqrx #920: Suggestion to change USB/LSB step size

pe1chl, 2020-12-27:
I suggest to change the frequency step size for USB and LSB from 100 Hz to 50 Hz.
This makes a noticable improvement when tuning signals for best voice quality.
I make this change every time I download the software, in file src/applications/gqrx/mainwindow.cpp:
    case DockRxOpt::MODE_LSB:
        /* LSB */
        rx->set_demod(receiver::RX_DEMOD_SSB);
        ui->plotter->setDemodRanges(-40000, -100, -5000, 0, false);
        uiDockAudio->setFftRange(0,3000);
        click_res = 50;
        break;
(same for USB)

Of course even better would be to have some setup of custom step size per mode in the user interface but that would involve more work.

argilo, 2023-04-16:
I had a look back at the commit history to see how the tuning step has evolved. Originally the tuning step was 10 Hz for all modes. Then a72891b28cf57936c19f2ffcf35d2c3b5ed3c046 (included in version 2.2) implemented per-mode tuning steps:

* Demod Off - 100 Hz
* AM - 100 Hz
* SSB - 10 Hz
* CW - 10 Hz
* Narrow FM - 100 Hz
* WFM - 1000 Hz

And finally, SSB was changed to 100 Hz and Demod off to 1000 Hz in c91693317c2cac00452b6cb6fbd4e1a40c0bc250 (included in version 2.4). I'm not sure what motivated the change, but perhaps 10 Hz was too low to allow convenient tuning with the mouse wheel. It also seems there was a preference to use powers of 10, which have the benefit of only changing one frequency digit. (I'm not sure this is important, though.)

Reducing the SSB step to 50 Hz seems like it might be a reasonable choice since it's not too much lower than the long-standing 100 Hz step, but it still allows a tuning error of ±25 Hz which is quite noticeable to my ear. What do you think about going back to the old 10 Hz step?

I'm not convinced that making the tuning step user-configurable is worth the added complexity, so my inclination is to either reduce the tuning step (to 50 or 10 Hz) or leave it untouched.

vladisslav2011, 2023-04-17:
Sometimes I want 10Hz SSB tuning step, but more often I want 500Hz SSB tuning step.
I'll make it user-configurable...

argilo, 2023-04-17:
> I'll make it user-configurable...

By the way, there is an open pull request for that: #928

As well as one to change the step size to 50 Hz: #1168

vladisslav2011, 2023-04-17:
I want to make per-mode tuning step user-configurable, not only SSB.
But I should fix "change 10 files and write 20 methods to add a single checkbox" issue first...