gqrx-sdr/gqrx #922: bladeRF

cherchais, 2020-12-31:
I installed gqrx by pybombs,like this:
> pybombs -p bladeRF install bladeRF gr-iqbal gr-osmosdr gqrx

after the installation is complete, start gqrx and report an error:
```bash
lsh@ubuntu:~$ gqrx 
gr-osmosdr 46204cea (0.1.5git) gnuradio 3.8.1.0
built-in source types: uhd hackrf bladerf rfspace airspy soapy redpitaya 
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-62-g7a3f1516
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # c79c...ee13 FW v2.3.2 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
Segmentation fault (core dumped)
```
Obviously it detects my bladeRF, but there was a segfault, I don’t know how to solve it,Can anyone help me?

system version:ubuntu 18.04
bladeRF version:
```bash
lsh@ubuntu:~$ bladeRF-cli -i
bladeRF> version

  bladeRF-cli version:        1.8.0-git-3a411c87c2-dirty
  libbladeRF version:         2.2.1-git-3a411c87c2-dirty

  Firmware version:           2.3.2
  FPGA version:               0.11.0 (configured from SPI flash)

```

argilo, 2021-01-01:
I don't have any problems with the BladeRF, and I'm running similar versions. It appears from "-dirty" in the version numbers that you've made changes to the bladeRF library before building it. Could that be where the problem lies?

I'd suggest running gqrx inside gdb and looking at the backtrace when the segfault occurs.

I'd also suggest trying the AppImage version (available at https://github.com/csete/gqrx/releases/tag/v2.14.4) which includes a copy of libbladeRF that should be working correctly.

argilo, 2021-10-04:
I'll close this off for now, since I haven't heard anything back.