gqrx-sdr/gqrx #923: Soapy remote

human39, 2020-12-31:
I spent a good part of last evening trying to connect to a soapy remote instance on my local network with gqrx.

The connection string I've been trying, while set to "Other":

`soapy=0,remote=<ip>,driver=remote,driver:remote=sdrplay`

I have verified that SoapyRemote is working with CubicSDR, but I prefer gqrx's intuitive interface.  I'm using to the AppImage release on Ubuntu 20.04.  

I'm not seeing _any_ connection attempts on the SoapyRemote side of things.

Any guidance would be appreciated.

argilo, 2021-01-01:
This is happening because SoapyRemote is not packaged in the AppImage release of Gqrx. I'll have to figure out how to do that.

human39, 2021-01-01:
Thanks for the reply @argilo!

By the way, thanks for providing an AppImage!  Happy New Year!

argilo, 2021-01-02:
I added SoapyRemote to the AppImage build in a15ef13dac99d1207954a5f8cd6b211856168d6a.

@human39 I'd appreciate if you could test the pre-release version available here: https://github.com/csete/gqrx/actions/runs/457904245

closed 2021-01-02 by argilo