gqrx-sdr/gqrx #925: After update: error while loading shared libraries: libgnuradio-analog.so.3.8.1

BaumGuard, 2021-01-01:
Hello,

today I've installed a bunch of updates including gnuradio. But after installing the updates, gqrx doesn't start anymore. Launching it from the terminal, the following error message is being shown:
gqrx: error while loading shared libraries: libgnuradio-analog.so.3.8.1: cannot open shared object file: No such file or directory

I've tried to donwgrade and reinstall gnuradio and gqrx but it doesn't work either.

I'm using Manjaro and installed gqrx from Manjaro's official repository.
The AppImage works fine but not the version from the repository.

Thank you for your help in advance and have a lucky new year!

ZirFi, 2021-01-01:
I have same platform and problem.

argilo, 2021-01-01:
This sounds like a bug in Manjaro, not Gqrx.

argilo, 2021-01-01:
Closing, since this is almost certainly a packaging issue in Manjaro / Arch.

closed 2021-01-01 by argilo

BaumGuard, 2021-01-01:
And what can I do get it working again? Please don't be so hasty and close issues before other people may have had the chance to help me and ZirFi, who also has that problem!

argilo, 2021-01-01:
Please report this issue to the maintainer of the Arch package. There is nothing I can do about this problem.

shuyuan-liu, 2021-01-01:
@BaumGuard Try building from source. Not too difficult, and certainly will not have this problem.

argilo, 2021-01-02:
Another option is to use an AppImage release: https://github.com/csete/gqrx/releases