gqrx-sdr/gqrx #929: GQRX crashes often with tcp over wifi

hirve, 2021-01-04:
Server is Raspberry PI 3, client is Ubuntu, Gqrx 2.11.5

```
SSSSSSSSSSSSSterminate called after throwing an instance of 'std::runtime_error'
  what():  SoapyRPCUnpacker::recv() FAIL test server connection: connect(192.168.2.227:55132) [110: Connection timed out]
Aborted (core dumped)
```

Maybe there is any option to make timeout infinite?

LongnoseRob, 2021-01-09:
I would recommend to have a look at this part of the SoapyRemote documentation:
https://github.com/pothosware/SoapyRemote/wiki#the-remotetimeout-key

hirve, 2021-01-11:
Tried but it didn't help, the init line is:
`driver=remote,remote=tcp://192.168.2.227:55132,remote:driver=airspyhf,serial=3652fc5d47cb39f7,soapy=1,remote:timeout=3000000`
I understand 3000000 is a 3 sec timeout, but GQRX crashes in some about 1 sec after changing frequency and server is continuing working after that. Maybe something wrong in the init line?

hirve, 2021-01-14:
Found workaround by taking IQ stream on the server and piping it with netcat to GQRX. GQRX sees it like an IQ file. So the problem is solved for me, feel free to close it if you like.